cra-ros-pkg/robot_localization #489: filtered odometry overestimation

abhileshborode, 2019-09-18:
**Required Info:**

- Operating System:
  - Ubuntu 16.04) -->
- Installation type:
  - 

#### Expected behavior
If I drive 10 m straight My filtered odometry should tell me I traveled 10 m straight

#### Actual behavior
If I drive 10 m straight My filtered odometry tells me I traveled 13.5 m straight

So I'm using robot_localization package for localization, with wheel encoders+IMU (Phiget / Realsense IMU) for fusion. The filtered odometry I'm observing is being overestimated by around 15% of that of the wheel odometry. If I drive 10 m straight My filtered odometry tells me I traveled 13.5 m straight.

I think this is happening due to the differential parameter being set to true for odom0 which is estimating velocity by calculating the time difference (correct me if I'm wrong). But if I set that param to false the entire filtered odometry is messed up. 

I tried using both the ekf/ukf localization nodes from the ros package. Here are my parameters: 

Here is the image of base_odom_raw(yellow) and filtered/odometry (red):
[image]

Additional information

> frequency: 20
sensor_timeout: 0.1
two_d_mode: true
transform_time_offset: 0.0
transform_timeout: 0.0
print_diagnostics: true
debug: false
debug_out_file: /path/to/debug/file.txt
publish_tf: true
 Whether to publish the acceleration state. Defaults to false if unspecified.
publish_acceleration: false
map_frame: world              # Defaults to "map" if unspecified
odom_frame: odom            # Defaults to "odom" if unspecified
base_link_frame: base_link  # Defaults to "base_link" if unspecified
world_frame: odom           # Defaults to the value of odom_frame if unspecified
odom0: /burro_base/odom
odom0_config: [true, true, false,
               false, false, false,
               false, false, false,
               false, false, false,
               false, false, false]
odom0_queue_size: 10
odom0_nodelay: true
odom0_differential: true
odom0_relative: false
odom0_pose_rejection_threshold: 250
odom0_twist_rejection_threshold: 250
imu0: /imu/data_raw
imu0_config: [false, false, false,
              false,  false,  false,
              false, false, false,
              false,  false,  true,
              false,  false,  false]
imu0_nodelay: true
imu0_differential: false
imu0_relative: false
imu0_pose_rejection_threshold: 250
imu0_twist_rejection_threshold: 250
imu0_queue_size: 10

Could anyone suggest any possible approaches to resolve this issue. @ayrton04 
ROS answer link: [https://answers.ros.org/question/333340/filtered-odometry-overestimation/](url)
Thanks.

closed 2019-09-18 by abhileshborode

SteveMacenski, 2019-09-18:
@abhileshborode thanks, I'll follow up on your ticket

ayrton04, 2019-09-19:
Just a note that you should include sample messages from every sensor input in the ROS Answers question. Ordinarily, I'd ask that you wait for a response there before posting an issue here, but I've been behind on my ROS Answers efforts lately.