cra-ros-pkg/robot_localization #493: Robot_localization launch error

shairi34ahmed, 2019-09-27:
ERROR: cannot launch node of type [robot_localization/ekf_localization_node]: can't locate node [ekf_localization_node] in package [robot_localization].

I have seen this error before and i don't see any concrete remarks/solution of this problem. I am using this package on ros kinetic and ubuntu 16.04, but i am this issue. I couldn't find a clear solution about this problem. i will be very thankful if someone could tell how to cleanly install this and cope this problem...

SteveMacenski, 2019-09-27:
Please open a question on ROS answers

shairi34ahmed, 2019-09-27:
I already have but still no answer. Here is the link of my question.
https://answers.ros.org/question/333411/error-cannot-launch-node-of-type-robot_localizationekf_localization_node/

SteveMacenski, 2019-09-27:
You have received a bunch of other people commenting trying to help (**far** more than I typically see for a question on ros answers in general), cross listing it here splits the conversation and makes folks less willing to help fix your issue which isn't with this codebase but your specific setup.

Please close this ticket

shairi34ahmed, 2019-09-27:
The reason i raised this question was i see many people have this issue, so there might be something, which requires not what you can say general settings or something else. But still i pursue my query on ros answers and close this ticket.

closed 2019-09-27 by shairi34ahmed