cra-ros-pkg/robot_localization #499: Navsat transform node start-up filtering?

GrantDare, 2019-10-29:
## Feature request

#### Feature description
Nick Charron mentioned in his [tutorial ](https://github.com/nickcharron/waypoint_nav) of the outdoor husky navigation that there were plans to filter start-up on the navsat transform node to provide a more accurate initialization (given GPS noise). I was wondering if this was implemented, planned, or trashed?

ayrton04, 2019-10-30:
If I'm being honest, I wouldn't expect to see a lot of new feature development on this package from me. We (Locus) are moving towards [fuse](https://github.com/locusrobotics/fuse) for our state estimation needs, and my free time for working on things that aren't directly required by work is pretty close to 0 these days. But I'll leave this here in the event that someone wants to take a crack at it.

GrantDare, 2019-11-05:
Thanks @ayrton04, fuse looks great! What is the status of the package?

ayrton04, 2019-11-07:
It should work pretty well for 2D robots. There's some work to be done to support robots that operate in 3D, and we don't yet have anything that handles GPS data. Stephen Williams is the one who designed and wrote the vast majority of the package. He's much better at state estimation than I am, so I think people will be pleased with the end result.

ThePapaG, 2019-11-07:
Unfortunately, my ultimate goal will require 3D localisation. Is there a timeline for the package or just _as it happens_?

ayrton04, 2019-11-18:
The latter, I'm afraid. We'd welcome PRs, though!

SteveMacenski, 2019-11-18:
(Easier to submit PRs with documentation, a public facing tickets/plan, and the 2D basic fuse-rl example to work back from. Don’t think I haven’t thought about it. I’d rather invest time into that than ROS2-RL but the barrier to entry is too high to guess what’s happening and how everyone plans to use a generic optimization framework with a design doc or roadmap of goals.)