cra-ros-pkg/robot_localization #500: Strange estimator behavior

mohradza, 2019-11-06:
Ubuntu 18.04.3 LTS. Forked the most recent version of the master branch on GitHub.

I am seeing some strange estimator behavior where the estimator seems to be stuck in a predict cycle for several seconds, even though it appears the sensor measurements are coming in consistently. 

Some more background: We are trying to generate odometry estimates from a single IMU and two "velocity sensors". These "velocity sensors" provide measurements of linear velocity in the body frame. During our initial testing, we were able to generate pretty accurate odometry estimates, over the course of a 120m walk with our sensor rig. Since original testing, we have moved all the sensors to a new vehicle. The biggest difference between the rig and the vehicle is the placement of the sensors, however we have adjusted the sensorX->base_link transforms to reflect their new locations. With the new setup, we keep getting the same results, which look like this:

[image]

This is a plot of the measured velocities vs the estimated velocity, where the measured velocities are shown in blue and green and the estimate in red. One can see that up to approximately the 10 second mark, the estimator is tracking appropriately. Then it just stops and has a seemingly linear slope. The position plots show the inconsistencies as well:

[image]

For completeness, here are the linear acceleration measurements coming from the imu:
[image]

So far we have tried making several changes. Changing the timeout frequency for the sensors did not have any effect. We have reduced the number of sensors down to a single velocity sensor, and the results are the same. We have tried changing the data rates of the sensors, which did not have any effect either. The issue seems to be related to specific motions. Here are the velocity plots from a different dataset, where the vehicle motions are the same as previous datasets:

[image]

If anyone has any insight as to what might be causing this behavior and what I need to do to fix it, please let me know! I can provide bagfiles for these datasets and the associated launch files if that would help debugging efforts.

Cheers!

mohradza, 2019-11-06:
I am not sure if this is a bug necessarily, but I cannot figure out how to change the tag.

mohradza, 2019-11-06:
This appears to be a result of the rejection threshold being applied to the velocity measurements. After commenting out the thresholds, these blips in the estimates no longer exist.

closed 2019-11-06 by mohradza

ayrton04, 2019-11-07:
In the future, I'd recommend asking questions on answers.ros.org before submitting tickets. If you already did that, I apologise. I'm having a hard time answering questions in a timely fashion.