cra-ros-pkg/robot_localization #502: offset problem

Fanny-one, 2019-11-15:
@ayrton04  Hi, I have used robot_localization in my project. In my urdf my rpy of imu is {0 0 -1.6301}. And according to api I set offset 1.57. When I use SLAM and localization simultaneously, they always match badly. I think my slam is unproblematic and the parameter of localization is improper probably.
Now I hope your help, wait for your reponse.

ayrton04, 2019-11-15:
Hi, thanks for the question. Before we mark things as bugs, please ask questions on answers.ros.org. If you have already done so, please point me to it, and then I ask your patience. I am swamped with other things at the moment.

closed 2019-11-15 by ayrton04

Fanny-one, 2019-11-18:
Sorry, it' my fault.  I did not intend to mark it as a bugs, but I just want to ask for help.