cra-ros-pkg/robot_localization #503: parameter problem

Fanny-one, 2019-11-18:
Hi， I have used robot_localization in my project. In my urdf my rpy of imu is {0 0 -1.6301} as  show in the graph.
[image]
Besides, the yaw is from dual-antenna direction not magnetometer so I needn't magnetic_declination_ . According to api I set offset 1.57. When I use SLAM and localization simultaneously, they always match badly. I think my slam is unproblematic and the parameter of localization is improper probably. I don't know why.

[image]

Now I hope your help, wait for your reponse.

ayrton04, 2019-11-18:
Hi,

Thanks for the question. However, we ask that all questions be directed to answers.ros.org. If you've already asked this there, please provide the link. Thanks.

closed 2019-11-18 by ayrton04

Fanny-one, 2019-11-19:
@ayrton04  
https://answers.ros.org/question/337930/robot_localization-parameter/
Thanks very much!