cra-ros-pkg/robot_localization #504: Transformation automatic apply

florianspy, 2019-11-19:
My Question / Feature request is the following:
I get one topic with visual odometry data which is in camera_frame 
and one tropic with odometry data which is in odometry frame ,
so if the data gets fused can i directly use the topics or must they be transformed before?
This leads me to another question in which frame is the output of the ekf?

SteveMacenski, 2019-11-19:
@florianspy please read the docs. This is outlined pretty clearly in my opinion. If its unclear, we would welcome a PR to make more clear.

florianspy, 2019-11-20:
Thank you very much for your answer. I finally found the corresponding chapter in the documentation " All pose data (position and orientation) is transformed from the message header’s frame_id into the coordinate frame specified by the world_frame parameter (typically map or odom). In the message itself, this specifically refers to everything contained within the pose property. All twist data (linear and angular velocity) is transformed from the child_frame_id of the message into the coordinate frame specified by the base_link_frame parameter (typically base_link)."  
My question is what happens if the corresponding transform into world_frame from frame_id specified in the pose data is not found?
I tried it out today with giving only one data source with frame_id in pose called RGBDodom and world_frame called odom without giving the neccessary transform from odom to RGBDodom.
In the terminal running roslaunch robot_localization ekf_template.launch,
no error message is given.
While /odometry/filtered simply gives no output. Is this the expected behaviour?

SteveMacenski, 2019-11-20:
I’d say yes, thats my expected behavior

ayrton04, 2019-11-28:
The problem with printing out a warning or error every time a transform fails is that it can get spammy very quickly if the transform is dynamic. The method we use for looking up transforms _can_ print warnings, but [we have it defaulting to silent](https://github.com/cra-ros-pkg/robot_localization/blob/6956176041630361364c06f554e319988e251300/include/robot_localization/ros_filter_utilities.h#L80). I'd accept a PR that makes this a parameter, such that if you _want_ transform warnings, you get them.

florianspy, 2019-12-03:
Dear ayrton04,
Such a feature to enable/disable with parameter would be great.

SteveMacenski, 2020-03-23:
@florianspy are you going to submit a PR?

florianspy, 2020-04-09:
Sorry took me a while till i saw your feedback i create a pullrequest now

closed 2021-04-14 by ayrton04