cra-ros-pkg/robot_localization #507: Thread safety for measurementQueue_

zhangjian-neu, 2019-11-27:
Dear authors,

Recently, I studied the package robot_localization. I find the variable 'measurementQueue_' can be updated in different callback functions, when new measurement ROS topic comes. 

As I know, the "topic-callback" mechanism is concurrent. This means the variable 'measurementQueue_' may be **concurrently** updated, when multiple measurement ROS topics come. 

Without extra thread safety mechanism , currently modifying the variable 'measurementQueue_' is not safe.  I hope to inquire where this issue is dealt with in the source codes, or whether this issue will be improved in future work.

Really thanks for your effort! 

I am looking forward to hearing from you.

ayrton04, 2019-11-27:
> As I know, the "topic-callback" mechanism is concurrent

No, this is not accurate. The callback queue is single threaded, and callbacks are fired sequentially. See [this](http://wiki.ros.org/roscpp/Overview/Callbacks%20and%20Spinning) wiki for more information.

closed 2019-11-27 by ayrton04

zhangjian-neu, 2019-11-28:
Great! Sincerely thanks for your kindly reply.