cra-ros-pkg/robot_localization #520: Matrix size error in RosRobotLocalizationListener::getState()

mabelzhang, 2020-01-11:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - From source
- Version or commit hash:
  - melodic-devel ba4010aa44d532ac509cbeabc7b95c71e785cdc9

### Steps to reproduce issue

This bug was found while porting classes to Dashing. The test gets an Eigen run-time error in Dashing, but doesn't get an error in Melodic, for whatever reason, though the matrix size is just as wrong in Melodic.

The test `LocalizationListenerTest.GetStateOfRelatedFrame` in `test_ros_robot_localization_listener.cpp` exposes a matrix dimensions error in matrix multiplication in `ros_robot_localization_listener.cpp` that isn't caught by Eigen for some reason.

The offending line is https://github.com/cra-ros-pkg/robot_localization/blob/melodic-devel/src/ros_robot_localization_listener.cpp#L504

It multiplies a `rot_6d` matrix which is 6 x 6, by a covariance matrix which in the test is 15 x 15, by the transpose of the 6 x 6 matrix. I added these 4 lines to inspect the matrices:
```
std::cerr << rot_6d << std::endl;
std::cerr << estimator_state.covariance.eval() << std::endl;
std::cerr << (rot_6d * estimator_state.covariance.eval()) << std::endl;
std::cerr << (rot_6d * estimator_state.covariance.eval() * rot_6d.transpose()) << std::endl;
```

Then run the test:
```
$ rosrun robot_localization test_ros_robot_localization_listener_publisher
$ rosrun robot_localization test_ros_robot_localization_listener
```

This prints 4 matrices with size 6 x 6, 15 x 15, 6 x 15 (!!), 6 x 6 (!!).

### Expected behavior
Test fails. The covariance matrix's size depends on the `STATE_SIZE`, which is declared in `include/robot_localization/filter_common.h`:
```
const int STATE_SIZE = 15;
```

### Actual behavior
Test passes.

### Additional information

In Dashing, I'm not sure what is different, but running the test will actually result in run-time error from Eigen, which is actually correct:
```
test_ros_robot_localization_listener: /usr/include/eigen3/Eigen/src/Core/Product.h:97: Eigen::Product<Lhs, Rhs, Option>::Product(const Lhs&, const Rhs&) [with _Lhs = Eigen::Matrix<double, -1, -1>; _Rhs = Eigen::Matrix<double, -1, -1>; int Option = 0; Eigen::Product<Lhs, Rhs, Option>::Lhs = Eigen::Matrix<double, -1, -1>; Eigen::Product<Lhs, Rhs, Option>::Rhs = Eigen::Matrix<double, -1, -1>]: Assertion `lhs.cols() == rhs.rows() && "invalid matrix product" && "if you wanted a coeff-wise or a dot product use the respective explicit functions"' failed.
```

I don't know how important this class is. We are porting this and `robot_localization_estimator.cpp` to Dashing because they seemed important. But this error seems huge, given that it's in a function called `getState()`. I have not followed through all the math in the function to figure out what the intention is and how to fix it.

ayrton04, 2020-01-19:
I won't get the cycles to address this, I'm afraid. As reported in https://github.com/cra-ros-pkg/robot_localization/pull/521, I am not familiar enough with this node to rapidly address issues with it. Feel free to fix or leave this open. Sorry!

rokusottervanger, 2020-01-19:
@ayrton04 referred to me in #521, so following up here. 

TLDR: Here's a concise answer to implement the covariance transformation causing the error here: https://robotics.stackexchange.com/questions/2556/how-to-rotate-covariance Not sure whether this is entirely correct though...

Long story:
Fundamentally, I never trusted the idea behind transformation of a covariance matrix when linear translation is involved between the measured position and the point of which the state is estimated in the filter (a distance between sensor and base frame, basically). I've heard people argue that the translation can be ignored in this transformation, whereas it seems evident to me that an uncertainty in a measured position of a translated point would transform to uncertainty in the orientation of the estimated pose, and vice versa. That's why I never implemented the covariance matrix transformation, and left it open for someone else to fill in. Apparently, @reinzor did implement this, but I don't know the details of this. If he cannot help with this, I'll gladly step in to look into the issue to at least fix the code to do what is intended. But I'd have to make some more time for that than I have available ATM.

reinzor, 2020-01-20:
I agree with @rokusottervanger that these covariance transformation not always make sense. However, there is still a bug here (as reported). The intented behavior is to only transform the 6x6 upper left matrix ([as being done here](https://github.com/cra-ros-pkg/robot_localization/blob/ba4010aa44d532ac509cbeabc7b95c71e785cdc9/src/ros_filter.cpp#L2722)). However, in this case, we were not taking into account that the covariance matrix is indeed `15x15`. 

I just submitted a pull request.

mabelzhang, 2020-01-21:
Nice! Thank you for the quick responses and fix :D

SteveMacenski, 2020-01-21:
@mabelzhang can you add to one of your dashing PRs for ROS2

Merging for Melodic

closed 2020-01-21 by SteveMacenski

mabelzhang, 2020-01-21:
Added to #521. Test now passes in Dashing.