cra-ros-pkg/robot_localization #524: navsat_transform: Parameter 'wait_for_datum' causes change to UTM transform.

Karl60, 2020-01-19:
## Bug report
Selecting 'true' or 'false' for the navsat_transform_node parameter "wait_for_datum" changes the published utm transform between odom->utm and map->utm .

**Required Info:**

- Operating System:
  - Ubuntu 18.04

- Installation type: 
  - binaries
- Version or commit hash:
  - 

#### Steps to reproduce issue

- navsat_transform node config: 
```
    # Odometry input with frame_id: map. (From the ekf_se_map node, world_frame: map)
    broadcast_utm_transform: true
    broadcast_utm_transform_as_parent_frame: false
    datum: [56.00000, -3.00000, 0.0]
```
 - Change parameter "wait_for_datum" between True and False.  

#### Expected behavior:
UTM transform created from world-frame of subscribed odometry topic, ie map->utm.

#### Actual behavior
```
wait_for_datum: true  # we get an odom->utm transform.
wait_for_datum: false # we get a map->utm transform.
```

#### Additional information

RonaldEnsing, 2020-07-21:
I am running into this as well.

ayrton04, 2020-07-21:
Try this?

https://github.com/cra-ros-pkg/robot_localization/pull/576

RonaldEnsing, 2020-07-21:
Yes, #576 works for me.

RonaldEnsing, 2020-07-21:
Will #576 be merged to master?