cra-ros-pkg/robot_localization #531: ros2 dashing exception while running

kikass13, 2020-01-29:
## Bug report

**Required Info:**

- Operating System:
  - 18.04
- Installation type:
  - from source 
- Version or commit hash:
  - dashing-devel

#### Steps to reproduce issue
Launch the se-node and publish std_msgs/Imu data for the kalmann filter

#### Expected behavior
/odometry/filtered is published and the node works 

#### Actual behavior
the node crashes 
```
[se_node-5] terminate called after throwing an instance of 'std::runtime_error'
[se_node-5]   what():  can't subtract times with different time sources [1 != 2]
[ERROR] [se_node-5]: process has died [pid 2090, exit code -6, cmd '/sandbox/ros_ws_rl/install/robot_localization/lib/robot_localization/se_node --ros-args -r __node:=se_node --params-file /sandbox/ros_ws_rl/install/robot_localization/share/robot_localization/params/ekf.yaml -r /imu:=/imu/data'].
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
```

#### Additional information

The error occurs here:
https://github.com/cra-ros-pkg/robot_localization/blob/dashing-devel/src/filter_base.cpp#L357

configuring the ekf.yaml (which was loaded into the node via ros2 launch) with:
```    use_control: false    ```
solves this problem, because the mentioned time check is skipped.

We (@kikass13 @scoopySHI ) found this bug yesterday while integrating the robot_localization into our test vehicle.

SteveMacenski, 2020-03-23:
Please ask on ROS Answers (or look it up there, I'm pretty sure I've answered this before there) as this is a setup and not a bug in the code, most likely.

closed 2020-03-23 by SteveMacenski

QQting, 2020-04-16:
Hi @kikass13, I also met this issue today, could you advise me how you solve it? Thanks.

kikass13, 2020-04-16:
@QQting 

I did :)
the issue describes where in the code the problem is and how to skip that specific time check.

 @SteveMacenski said that this is not a bug, but a diff of two times with two different sources leads to an exception (ros2) which is not catched (at least not in the version I used when creating this issue) is a bug :)