cra-ros-pkg/robot_localization #535: Dashing: Tests fail on `arm64` hardware

JWhitleyAStuff, 2020-02-06:
## Bug report

**Required Info:**

- Operating System:
  - Docker `ros:dashing-ros-base` container (`arm64` version)
- Installation type:
  - From source
- Version or commit hash:
  - 0b5f69128ad0c364dfdf5e478e9abb6369a59743

#### Steps to reproduce issue
Build on `arm64` hardware or use Docker following the instructions at https://www.stereolabs.com/docs/docker/building-arm-container-on-x86/ to run on x86 hardware. You'll then have to enable experimental Docker features to pull the `arm64` ROS2 image using the instructions [here](https://github.com/moby/moby/issues/30585#issuecomment-280822231). Once that is done, run `docker run -it --rm --platform arm64 ros:dashing-ros-base` to get you into the container (if you already have this container in your native architecture, you'll have to delete it to get the `arm64` version). From inside the container or on native hardware, run:

```
$ uname -m (to make sure you're in the arm64 container or arm64 hardware)
$ mkdir -p ros2_ws/src && cd ros2_ws/src
$ git clone https://github.com/cra-ros-pkg/robot_localization -b dashing-devel
$ cd ..
$ apt-get update && apt-get install python-rosdep
$ rosdep init && rosdep update
$ rosdep install --from-paths src --ignore-src -y
$ colcon build
$ colcon test
$ colcon test-result --verbose
```

**Note:** In Docker with `binfmt`, building and testing may take 15+ minutes each.

#### Expected behavior
All tests pass

#### Actual behavior
The following errors are produced from `colcon test-result --verbose`:

```
build/robot_localization/Testing/20200206-2226/Test.xml: 41 tests, 0 errors, 3 failures, 0 skipped
build/robot_localization/test_results/robot_localization/test_ekf_localization_node_interfaces.xml: 1 test, 0 errors, 1 failure, 0 skipped
- robot_localization test_ekf_localization_node_interfaces.missing_result
  <<< failure message
    The test did not generate a result file.
  >>>
build/robot_localization/test_results/robot_localization/test_filter_base_diagnostics_timestamps.xml: 1 test, 0 errors, 1 failure, 0 skipped
- robot_localization test_filter_base_diagnostics_timestamps.missing_result
  <<< failure message
    The test did not generate a result file:
    
    [INFO] [launch]: All log files can be found below /root/.ros/log/2020-02-06-22-28-21-611983-debd42891711-16926
    [INFO] [launch]: Default logging verbosity is set to INFO
    [INFO] [ekf_node-1]: process started with pid [16936]
    [INFO] [test_filter_base_diagnostics_timestamps-2]: process started with pid [16939]
    [ekf_node-1] [WARN] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
    [test_filter_base_diagnostics_timestamps-2] [==========] Running 3 tests from 1 test case.
    [test_filter_base_diagnostics_timestamps-2] [----------] Global test environment set-up.
    [test_filter_base_diagnostics_timestamps-2] [----------] 3 tests from FilterBaseDiagnosticsTest
    [test_filter_base_diagnostics_timestamps-2] [ RUN      ] FilterBaseDiagnosticsTest.EmptyTimestamps
    [test_filter_base_diagnostics_timestamps-2] /ros2_ws/src/robot_localization/test/test_filter_base_diagnostics_timestamps.cpp:265: Failure
    [test_filter_base_diagnostics_timestamps-2] Value of: received_warning_imu
    [test_filter_base_diagnostics_timestamps-2]   Actual: false
    [test_filter_base_diagnostics_timestamps-2] Expected: true
    [test_filter_base_diagnostics_timestamps-2] /ros2_ws/src/robot_localization/test/test_filter_base_diagnostics_timestamps.cpp:266: Failure
    [test_filter_base_diagnostics_timestamps-2] Value of: received_warning_odom
    [test_filter_base_diagnostics_timestamps-2]   Actual: false
    [test_filter_base_diagnostics_timestamps-2] Expected: true
    [test_filter_base_diagnostics_timestamps-2] /ros2_ws/src/robot_localization/test/test_filter_base_diagnostics_timestamps.cpp:267: Failure
    [test_filter_base_diagnostics_timestamps-2] Value of: received_warning_twist
    [test_filter_base_diagnostics_timestamps-2]   Actual: false
    [test_filter_base_diagnostics_timestamps-2] Expected: true
    [test_filter_base_diagnostics_timestamps-2] /ros2_ws/src/robot_localization/test/test_filter_base_diagnostics_timestamps.cpp:268: Failure
    [test_filter_base_diagnostics_timestamps-2] Value of: received_warning_pose
[... 48 lines cut ...]
    [test_filter_base_diagnostics_timestamps-2] 
    [test_filter_base_diagnostics_timestamps-2] [----------] Global test environment tear-down
    [test_filter_base_diagnostics_timestamps-2] [==========] 3 tests from 1 test case ran. (9287 ms total)
    [test_filter_base_diagnostics_timestamps-2] [  PASSED  ] 0 tests.
    [test_filter_base_diagnostics_timestamps-2] [  FAILED  ] 3 tests, listed below:
    [test_filter_base_diagnostics_timestamps-2] [  FAILED  ] FilterBaseDiagnosticsTest.EmptyTimestamps
    [test_filter_base_diagnostics_timestamps-2] [  FAILED  ] FilterBaseDiagnosticsTest.TimestampsBeforeSetPose
    [test_filter_base_diagnostics_timestamps-2] [  FAILED  ] FilterBaseDiagnosticsTest.TimestampsBeforePrevious
    [test_filter_base_diagnostics_timestamps-2] 
    [test_filter_base_diagnostics_timestamps-2]  3 FAILED TESTS
    [ERROR] [test_filter_base_diagnostics_timestamps-2]: process has died [pid 16939, exit code 1, cmd '/ros2_ws/install/robot_localization/lib/robot_localization/test_filter_base_diagnostics_timestamps'].
    [INFO] [ekf_node-1]: sending signal 'SIGINT' to process[ekf_node-1]
    [ekf_node-1] [INFO] [rclcpp]: signal_handler(signal_value=2)
    [INFO] [ekf_node-1]: process has finished cleanly [pid 16936]
  >>>
build/robot_localization/test_results/robot_localization/test_ukf_localization_node_interfaces.xml: 1 test, 0 errors, 1 failure, 0 skipped
- robot_localization test_ukf_localization_node_interfaces.missing_result
  <<< failure message
    The test did not generate a result file.
  >>>

```

#### Additional information
This package is being used by [the Autoware.Auto project](https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto), which requires support on `arm64` hardware.

JWhitleyWork, 2020-02-18:
@SteveMacenski I can't assign but I'm assuming you'll be the one to work on this.

SteveMacenski, 2020-02-18:
@JWhitleyWork I wasn't planning on it - as I do not have an ARM64 processor to test on

JWhitleyWork, 2020-02-18:
@SteveMacenski - I provided instructions in the issue description for running an `arm64` Docker container on `amd64` hardware.

SteveMacenski, 2020-02-18:
I’m also pretty underwater with other tasks. I can put this on my queue, but it’ll likely be at minimum 6 weeks before I’ll get around to it.

This test I think is the same failing in the build farm that Mabel at least briefly looked at

mabelzhang, 2020-02-18:
Ah, I came to look at this thread after that build farm failure. Heh... this gives me more information to test. I'll try something.

mabelzhang, 2020-02-25:
I followed the instructions above to set up the aarch64 Docker environment and am getting a lot of run-time printouts like:
```
[ekf_node-1] Unsupported setsockopt level=0 optname=32
```
It looks like a QEMU problem? @JWhitleyWork Have you run into this and happen to know how to set up the environment correctly to bypass this? My host machine is:
```
$ uname -m
x86_64
```

JWhitleyWork, 2020-04-01:
@mabelzhang I did see these errors but I don't believe they stopped me from compiling.

SteveMacenski, 2021-09-15:
Dashing is EOL

closed 2021-09-15 by SteveMacenski