cra-ros-pkg/robot_localization #546: Static drift at initial position

CrazyPopLin, 2020-03-20:
I am using a rosbag inclusing: gps data, filtered imu(using imu_filter_madgwick), imu raw data, DVL data(measure vehicle velocity), gps ekf odometry(the ground truth). I am trying to fuse imu-DVL offline to compare the gps ekf odometry.

[image]
**Figure 1: gps fused path and filtered imu fused path**

In Figure 1, the green is the gps-fused path. white is the imu-filtered and DVL fused path. I set a _initial_state_ for white path, so that two paths were started from some location. 

[image]
**Figure 2: gps fused path and offline filtered imu fused path**

In Figure 2, I re-do  _imu_filter_madgwick_  offline for the imu raw data, and take this filtered data for ekf fuse. I am using exactly the same setup. The only difference is the imu input for ekf. The Fig.1 using from rosgbag, the Fig.2 using from offline processed. Basically, those two filtered imu data are same thing. There is always have a drift about 99.2 degree at the beginning.

The reason I re-do filter imu raw data, because I try different imu parameters for ekf. But here, I am using the same imu parameters.

Any one can give me some suggestions. Thanks.

ayrton04, 2020-03-23:
Thanks for the question. However, all questions should be directed to answers.ros.org. If you've already posted the question there, I'd ask for your patience, as I am extremely busy these days and have less and less time to devote to this package.

In any case, feel free to add the link to the ROS Answers question, whenever it gets/was posted.

closed 2020-03-23 by ayrton04

CrazyPopLin, 2020-05-29:
hi, I've post a ROS Answers question here: https://answers.ros.org/question/353573/weird-initial-path-from-robot_localizations-ekf-node/
Please give me some suggestions and thanks for your help.