cra-ros-pkg/robot_localization #555: Failed to build dependent package.

blacksoul000, 2020-04-15:
**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - binaries and from source
- Version or commit hash:
  - 700ea7efb4a7e38c14a343a541f5d738453f0d5f

#### Steps to reproduce issue
ros2 pkg create --build-type ament_cmake --node-name test_node test_package --dependencies robot_localization
colcon build

#### Expected behavior
Dependent package builds fine.

#### Actual behavior
CMake Error at /opt/ros/dashing/share/robot_localization/cmake/ament_cmake_export_libraries-extras.cmake:48 (message):
  Package 'robot_localization' exports the library 'rl_lib' which couldn't be
  found
Call Stack (most recent call first):
  /opt/ros/dashing/share/robot_localization/cmake/robot_localizationConfig.cmake:38 (include)
  CMakeLists.txt:19 (find_package)

SteveMacenski, 2020-04-15:
PR welcome

SteveMacenski, 2020-07-20:
Resolved in 3.1.1

closed 2020-07-20 by SteveMacenski

shlokgoel, 2021-01-14:
Hi @SteveMacenski I am getting the same error when trying to install ros1_bridge. 

[image]
For reference: 

I am trying to install ros1_bridge inside ade. I have installed ros-dashing using Autoware.Auto and ros-melodic-desktop-full. 

Do you know why I am getting this error? 

Thanks!