cra-ros-pkg/robot_localization #558: Release robot_localization into ROS Noetic?

sloretz, 2020-04-23:
Would you mind releasing `robot_localization` to ROS Noetic? It looks like all of its dependencies have been released, and recursively 15 repos need it.

I'm not sure how much work it will need. The [Noetic Migration Guide](http://wiki.ros.org/noetic/Migration) or the [guide about transitioning ROS packages to Python 3](http://wiki.ros.org/UsingPython3/) may be useful.

SteveMacenski, 2020-04-23:
I can do this, shouldn’t be much effort. There’s an active question in #557 on if changing the install locations would break ABI that’s my only blocker as a fork in the road about direction to go.

Timple, 2020-05-22:
The [call for testing](https://discourse.ros.org/t/ros-noetic-call-for-testing) reveiled that this package was not yet released for noetic.
Now #557 is merged, would a release be in order?

SteveMacenski, 2020-05-22:
@ayrton04 I went to release and I get a warning about potential items to address and patches on various branches. I haven't made use of the bloom patches capabilities (are those documented anywhere?) so that makes me wary to doing it myself. Can you release for noetic on `melodic-devel`?

ayrton04, 2020-05-26:
I am happy to bloom it. As I recall, I had to bloom, then manually add an `-O3` build flag somewhere in a release repo, and then bloom again. I'll sort it out early next week.

ayrton04, 2020-06-01:
We may need to branch for noetic after all. This rosdep key doesn't exist for focal:

https://github.com/cra-ros-pkg/robot_localization/blob/melodic-devel/package.xml#L38

Timple, 2020-06-01:
You can have conditional dependencies for package format 3. Which is available at least as of kinetic.

ayrton04, 2020-06-03:
So I've not used the conditional dependencies before (clearly), but looking at the documentation, it looks like I need to be able to do this:

```
<build_depend condition="foo">python-catkin-pkg</build_depend>
<build_depend condition="bar">python3-catkin-pkg</build_depend>
```

I'm sure I'm missing something obvious, but what conditions would we use in this case? I need to know what the name of that dependency is for the platform on which we're building. The build farm builds for more than just Ubuntu, so I need a condition that will work across platforms. If they all have the same dependency name, then I can do this:

```
<build_depend condition="$ROS_DISTRO != noetic">python-catkin-pkg</build_depend>
<build_depend condition="$ROS_DISTRO == noetic">python3-catkin-pkg</build_depend>
```

^ That at least functions on focal. If it works on the other Noetic [target platforms](https://www.ros.org/reps/rep-0003.html#noetic-ninjemys-may-2020-may-2025), then it should behave. I'll PR that change and see what the build farm says for melodic.

SteveMacenski, 2020-06-03:
An example:
`<exec_depend condition="$ROS_PYTHON_VERSION == 2">python-numpy</exec_depend>`

ayrton04, 2020-06-03:
Ah, that's a much better env var to use. Thanks.

ayrton04, 2020-06-03:
https://github.com/ros/rosdistro/pull/25297

Once that gets merged and the buildfarm picks this up and (hopefully) builds it, I'll close this.

ayrton04, 2020-06-05:
This is now complete. Note that we're tracking a different [issue](https://github.com/cra-ros-pkg/robot_localization/issues/574) with `tf2`, but I think the fix is on their end in this case.

closed 2020-06-05 by ayrton04