cra-ros-pkg/robot_localization #562: First-Order Robot Localization

ewerlopes, 2020-04-29:
Hi,

I wonder whether it is possible to use `robot_localization` as a first-order approximator. In other words, whether it is possible to have it modeling just position and velocities without acceleration (making up to 12 variables for the state). It' seems I cannot find instructions for that in the documentation.

Thanks for the repo!
Best.

ewerlopes, 2020-04-29:
Hi again;

I just saw that, as a response to #8, you have added the acceleration data into the master branch for release 1.1.6. As I can see the current release is still backwards-compatible in the sense that if I provide a 12-D vector as message input config the code doesn't break but prints `Configuration vector for imu0_config should have 15 entries` instead.

Can you please confirm whether there is a way to not use acceleration at all, just position and velocity? Of course I can fallback to release 1.1.5 and use that version instead, but I don't know whether I would lose any relevant feature present in newer versions.

Thanks.

SteveMacenski, 2020-05-07:
Please ask on ROS Answers, but you can turn off acceleration from being fused with the configurations files. there's not a way to reformat the EKF at this time to completely remove it

closed 2020-05-07 by SteveMacenski