cra-ros-pkg/robot_localization #564: [ROS2] Replace publishers with msg references with unique_ptrs

SteveMacenski, 2020-05-02:
Right now, all the publishers use the message reference API for publishing information. If using intra-process components, that will incur a copy. For inter-process it adds jitter to the system. While for messages of this size it is probably not going to make a large system-level impact, it is better practice and may be good samples for future folks and 0.1% improvement for an hour's work is a pretty reasonable tradeoff. 

Ref: https://github.com/SteveMacenski/ros2_ouster_drivers/pull/41 on discussion and also how to change to the other API with `std::move`.

Sarath18, 2020-05-06:
I would like to take up this issue

SteveMacenski, 2020-05-06:
Go for it!

closed 2020-05-07 by SteveMacenski