cra-ros-pkg/robot_localization #574: TF_REPEATED_DATA warning in Noetic

ayrton04, 2020-06-04:
I'm running the package test suite in Noetic, and I am seeing this getting spammed:

```
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame base_link at time 1432235782.066014 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.1/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame base_link at time 1432235782.433942 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.1/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame base_link at time 1432235782.865701 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.1/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame base_link at time 1432235782.965662 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.1/src/buffer_core.cpp
```

This is clearly a new warning, and I'm assuming it's due to [this](https://github.com/cra-ros-pkg/robot_localization/issues/336#issuecomment-629725698) common situation:

- Cycle 1: receive measurement from sensor with time stamp `t1`. Fuse measurement. Publish state at time `t1`.
- Cycle 2: receive measurement from sensor with time stamp `t2` < `t1`. Rewind filter, insert measurement from `t2`, fuse measurement from `t2` and then measurement from `t1`. Publish state at time `t1`.

It looks like this helpful change was added, in which `tf2` would just overwrite the old transform with the new one:

https://github.com/ros/geometry2/commit/eefb50935bfd28223c87b3d708e916f7bcc4b4ca

However, that was later changed, such that attempting to broadcast the same transform at the same time would fail and spit out a warning:

https://github.com/ros/geometry2/commit/04625380bdff3f3e9e860fc0e85f71674ddd1587#diff-c25683741e1f7b4e50eb6c5cdcad9661R275

@SteveMacenski, what are your thoughts here? Should I open a ticket on the `geometry2` repo?

ayrton04, 2020-06-04:
Started a discussion here:

https://github.com/ros/geometry2/issues/467

SteveMacenski, 2020-06-04:
My feeling is that change in geometry2 was well intended, but shouldn't have been merged. That's a problem as time varying transformations in the presence of uncertainty may need to be updated. I wish TF2 also propagated covariance matrices.... maybe in TF3 ;-)

reobaird, 2020-09-18:
Any updates on a timeline for fixing this? It's really causing a lot of noise for us and is making debugging / logging other stuff more difficult.

ayrton04, 2020-09-18:
So my simple solution for this will be to add a parameter that allows the filter to repeat publication time stamps. However, I feel like that parameter should default to *false* for Noetic, but everything before Noetic should retain the old behavior for compatibility. @SteveMacenski, I'm thinking we're going to need a `noetic-devel`.

ayrton04, 2020-09-18:
@reobaird, give that branch a spin, if you have any cycles. Waiting to see if the warnings go away on the ROS buildfarm.

SteveMacenski, 2020-09-18:
I'm still not sure what the right answer is here. There's still discussions in the TF2 ticket. I don't want to be inflexible, but this feels to me like something that should be reverted given the pushback from the community in that ticket.

I wouldn't want to merge anything here creating what I think is "bad" behavior until we have some resolution on the TF2 ticket

acarrillo, 2021-01-08:
Howdy! Did this update to `robot_localization` ever make it out of the `ros-testing` apt repo? The stable `http://packages.ros.org/ros/ubuntu` repo seems to hold `2.6.8` as its latest version, whereas `ros-testing` has `2.7.1`:

```
acarrillo /opt/ros/noetic/share/robot_localization $ apt-cache policy ros-noetic-robot-localization 
ros-noetic-robot-localization:                                                                      
  Installed: 2.6.8-2focal.20201207.153320                                                           
  Candidate: 2.7.1-1focal.20201217.183251                                                           
  Version table:                                                                                    
     2.7.1-1focal.20201217.183251 500                                                               
        500 http://packages.ros.org/ros-testing/ubuntu focal/main amd64 Packages                    
 *** 2.6.8-2focal.20201207.153320 100                                                               
        100 /var/lib/dpkg/status                                                                    
acarrillo /opt/ros/noetic/share/robot_localization $                                                
```

Let me know if you need any help promoting the newest version to stable, my team really needs this feature! Our console logs are really degraded without it...

ayrton04, 2021-01-08:
I bloomed everything. I don't know if they did a sync yet or not.

ayrton04, 2021-01-08:
Looks like the last Noetic sync was on the 30th of November.

https://discourse.ros.org/t/new-packages-for-noetic-2020-11-30/17641

I bloomed the package in mid-December, so you'll just need to wait for the Open Robotics folks to do the sync.

closed 2021-04-14 by ayrton04