cra-ros-pkg/robot_localization #580: [ROS2] cppcheck fails on smart pointer macro inclusion

SteveMacenski, 2020-07-20:


ayrton04, 2020-07-21:
I'll get my ROS2 environment set up and have a run at this.

ayrton04, 2020-07-21:
Looks a lot like this issue:

https://github.com/ament/ament_lint/issues/116

ayrton04, 2020-07-21:
So here's the command the build farm is running:

```
Test command: /usr/bin/python3 "-u" "/opt/ros/foxy/share/ament_cmake_test/cmake/run_test.py" "/tmp/ws/test_results/robot_localization/cppcheck.xunit.xml" "--package-name" "robot_localization" "--output-file" "/tmp/ws/build_isolated/robot_localization/ament_cppcheck/cppcheck.txt" "--command" "/opt/ros/foxy/bin/ament_cppcheck" "--xunit-file" "/tmp/ws/test_results/robot_localization/cppcheck.xunit.xml" "--language" "c++"
```

Boiling that down to this, and running it on the file in question, yields the test failure we're seeing:

```
ament_cppcheck --language "c++" src/robot_localization/include/robot_localization/filter_base.hpp 
[src/robot_localization/include/robot_localization/filter_base.hpp:89]: (error: unknownMacro) There is an unknown macro here somewhere. Configuration is required. If RCLCPP_SMART_PTR_DEFINITIONS_NOT_COPYABLE is a macro then please configure it.
```

However, if I just include the `foxy` root include directory, it's fine:

```
ament_cppcheck --include_dirs /opt/ros/foxy/include --language "c++" src/robot_localization/include/robot_localization/filter_base.hpp 
No problems found
```

So at the moment, it's not looking in the root ROS2 include directory when it runs `ament_cppcheck`, but that seems odd, because surely lots of packages use those smart pointer definitions.

SteveMacenski, 2020-07-21:
(should we maybe just remove it and add the specific pointer definitions needed. 1 line replaced with 2 lines isn't that big of a deal to me and it just ends this problem)

ayrton04, 2020-07-21:
Yeah, I'll do that tomorrow. If you want to fix it right now, feel free, but the build farm is at least happy for now.

SteveMacenski, 2020-07-21:
I'll wait for you, I wouldn't want to take away your opportunity to dust off your ROS2 environment ;-) 

_One of us. One of us. One of us._

closed 2020-07-23 by ayrton04