cra-ros-pkg/robot_localization #584: Failed to meet update rate!

ibrahiminfinite, 2020-08-03:
Hi @ayrton04 ,

I am fairly new to the robot_localization package ,
I have set up my launch file based on provided example --> my  [file ](https://gist.github.com/ibrahiminfinite/97a82f0ce331601af3f913488944a666)

On launching the file i get the warning

>  " Failed to meet update rate !"

I found that the warning is from the function

> void RosFilter<T>::periodicUpdate

However , I do not understand the cause for this warning , if it is because of some parameter i have set or not.
Any advice is appreciated.

Thanks

SteveMacenski, 2020-08-03:
Please ask on ROS Answers

closed 2020-08-03 by SteveMacenski

mjs973, 2020-10-12:
If this is still a problem for you, take a look at issue #601 to see if it applies to your case. A code fix has been submitted for melodic.

kc944201, 2020-12-02:
@mjs973 HI, 
When I use ukf_localization_node to combine odometry and IMU data
the warning messages still logging on terminal

Could you help me solved this problem? 
Thank you

ayrton04, 2020-12-02:
Please direct all questions to answers.ros.org. If you have asked there, please be patient, and I will get to it as soon as I can.

JStech, 2021-05-26:
Sorry about that--it's [here](https://answers.ros.org/question/358574/robot_localization-failed-to-meet-update-rate-warning/).

SteveMacenski, 2021-05-26:
100 hz should be easy, are you building with release flags?

JStech, 2021-05-26:
Yes, ish: `RelWithDebInfo`

SteveMacenski, 2021-05-26:
I looked at some old files from a hobby robot I had a home equipped with a 4th gen i5 and I was running at at 80hz with no output errors, so I think that might be the case. It's unlikely that additional 20% rate would really impact things -- especially with 4 generations later's i7 :wink:

mjs973, 2021-05-26:
ekf_localization is written as a single threaded node; it's just going to blow up if you try to invoke the callbacks from multiple threads.

Have you looked at the --hz and --rate options to rosbag play?

ayrton04, 2021-05-27:
I'm actually wondering if this could be related to the measurements themselves.

If the timestamps on the measurements are not well synchronized, then at some cycle, you might get into a situation where:

- Some measurement comes in with an old timestamp
- The filter has to then rewind its history, insert that measurement, and then integrate all the other measurements up until the most recent measurement (where recency is determined by message timestamps)

I am also curious about this line:

https://github.com/cra-ros-pkg/robot_localization/blob/67098c2341b5d1ccbcceb8eede60e79db74814a6/src/ros_filter.cpp#L1864

The math is `(event.current_real - event.last_expected).toSec()`. Looking at the documentation, I see this:

```
ros::TimerEvent

ros::Time last_expected
In a perfect world, this is when the previous callback should have happened

ros::Time last_real
This is when the last callback actually happened

ros::Time current_expected
In a perfect world, this is when the current callback should have been called

ros::Time current_real
When the current callback is actually being called (ros::Time::now() as of immediately before calling the callback)

ros::WallTime profile.last_duration
Contains the duration of the last callback (end time minus start time). Note that this is always in wall-clock time.
```

I feel like we want to be concerned with how long each loop cycle _actually_ took. To that end, I wonder if we should change the math from `(event.current_real - event.last_expected).toSec()` to `(event.current_real - event.last_real).toSec()`, or even just use `profile.last_duration`. The latter would allow you to play back the bags very slowly, but still capture issues that are really down to being CPU bound.

mjs973, 2021-05-28:
@ayrton04 I should have time to look at your idea this weekend.

JStech, 2021-05-30:
> Have you looked at the --hz and --rate options to rosbag play?

Yes, running with `--rate 0.8` prevents the warnings.

mjs973, 2021-05-30:
> running with --rate 0.8 prevents the warnings.

That's gives us a hint, but it's hard to see how you are hitting a real-time limit since you said the cpu usage is low.

What happens if you try --hz 1000 with no --rate option? I'm wondering if rosbag play is sending bursts of messages?

JStech, 2021-05-30:
Running with `--hz 1000` (and no `--rate`) doesn't prevent the warnings but does change the behavior. Before they were printed at pretty regular intervals (about 2Hz) throughout the length of the bag. Now, there are very few warnings printed at the start of the bag (it's about 0.2Hz), but then it goes back to about 2Hz by the end.

I checked the bag in rqt_bag, and it does have some clumped IMU and GPS messages, although the header timestamps look more regular, so I might try to reindex the bag by timestamp (I admit I'm not 100% sure what I'm talking about here). Maybe this all is because of shoddy input data :blush:.

mjs973, 2021-05-31:
@ayrton04 I have thought about your proposal. The current check in periodicUpdate() allows for small, infrequent excursions above the 1*period duration without triggering the warning. For an edge-case example, if the filter's average update time is 0.95*period, with a few 1.20*period occurances, there will be no warning. Your suggestion will require lowering the threshold to 1*period, and that will print warnings a lot more I suspect.

My vote is to not change this code.

mjs973, 2021-05-31:
@JStech I pulled up the source code for rosbag play to see exactly what `--hz` does in melodic, and it affects only the rate of publishing to /clock. So given that, there's no reason it'd help your issue.

What update rate are you using for the ekf filter? How many topics are present in the rosbag file?

JStech, 2021-05-31:
I'm running two EKFs, each at 100Hz. There are three topics in the bag: GPS, IMU, and wheel odometry.

I rewrote the bag with the header timestamps (a la [this tutorial](http://wiki.ros.org/rosbag/Cookbook#Rewrite_bag_with_header_timestamps)), which also didn't help.

mjs973, 2021-05-31:
@JStech Ah, I forgot you told us that in your original post. I'm out of ideas.

As background, this warning is from a ros timer thread that is trying to run at the filter update rate, but something is preventing it from doing so. The most common cause of this is too much time being spent processing incoming message(s).

mjs973, 2021-06-01:
@JStech Do you still get the warning if you run just one ekf filter? Are you using nodes or nodelets?

JStech, 2021-06-02:
Yes, still happens with one filter. I'm running nodes, not nodelets.

mjs973, 2021-06-11:
@JStech While chasing down a bug unrelated to yours, I'm reminded that ros connections default to having the tcp transmit buffer enabled. Have you used the ekf private parameters to disable tcp buffering for your high message rate topics going into ekf? You can search for:

`_nodelay`

in ros_filter.cpp. The parameter names have the format `topicName + "_nodelay"`, and the value is a boolean. You want to set this to true.

mjs973, 2021-06-11:
@JStech Clarification: use the ekf's node's "local" name for the input, not the **actual** topic name e.g. odom0_nodelay = true

JStech, 2021-06-13:
Thanks for the suggestion--I tried setting all the `*_nodelay` values to `true` but the behavior didn't change.

ayoubasritii, 2023-04-11:
Hello @JStech @mjs973 
I'm facing the same warning messages and I have exactly the same setup, two EKF nodes with IMU and wheel odometry at 100 Hz. only 4Hz for GPS.
I'm interested in any new information on this subject.
Thank you.