cra-ros-pkg/robot_localization #587: Does the covariance is related to the velocity and acceleration?

improve100, 2020-08-24:
As I accelerate, I find that the covariance is related to my acceleration and my velocity.
I check the code found this:

double dFx_dP = (xCoeff * xVel + yCoeff * yVel + zCoeff * zVel) * delta +
                    (xCoeff * xAcc + yCoeff * yAcc + zCoeff * zAcc) * oneHalfATSquared;
    
double dFR_dP = (cpi * cpi * sr * pitchVel + cpi * cpi * cr * yawVel) * delta;

transferFunctionJacobian_(StateMemberX, StateMemberPitch) = dFx_dP;

It's say that the Jacobian matrix is related to the velocity and acceleration. 
When the velocity increases, the covariance of the entire EKF will increase, not only related to the time?
If I set a threshold of covariance, when I go too fast, I'm going to report an error very quickly.

Timple, 2020-08-24:
Cross post: https://answers.ros.org/question/360182/does-the-covariance-is-related-to-the-velocity-and-acceleration/

ayrton04, 2020-08-24:
I've been behind on my ROS answers attentiveness, but I'll try to catch up this week. Questions should only go there, thanks.

closed 2020-08-24 by ayrton04