cra-ros-pkg/robot_localization #590: CMake error when catkin_make

ANDUES, 2020-09-07:
When I build the robot_localization package from source, I got a CMake Error like this:

CMake Error at robot_localization/CMakeLists.txt:35 (find_package):
  By not providing "FindGeographicLib.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "GeographicLib", but CMake did not find one.

  Could not find a package configuration file provided by "GeographicLib"
  with any of the following names:

    GeographicLibConfig.cmake
    geographiclib-config.cmake

  Add the installation prefix of "GeographicLib" to CMAKE_PREFIX_PATH or set
  "GeographicLib_DIR" to a directory containing one of the above files.  If
  "GeographicLib" provides a separate development package or SDK, be sure it
  has been installed.

ANDUES, 2020-09-07:
I used rosdep install --from-paths src --ignore-src -r -y solved the problem thx

closed 2020-09-07 by ANDUES

narutojxl, 2020-09-24:
Hi @ANDUES, my OS is ubuntu16.04 and kinetic,  after `rosdep install --from-paths src --ignore-src -r -y `, then `catkin_make`:
> 
	CMake Error at /usr/share/cmake-3.17/Modules/FindPackageHandleStandardArgs.cmake:164 (message):
	  Could NOT find GeographicLib (missing: GeographicLib_LIBRARY_DIRS
	  GeographicLib_INCLUDE_DIRS)
	Call Stack (most recent call first):
	  /usr/share/cmake-3.17/Modules/FindPackageHandleStandardArgs.cmake:445 (_FPHSA_FAILURE_MESSAGE)
	  /usr/share/cmake/geographiclib/FindGeographicLib.cmake:28 (find_package_handle_standard_args)
	  robot_localization/CMakeLists.txt:35 (find_package)
	-- Configuring incomplete, errors occurred!
the line 28 of  /usr/share/cmake/geographiclib/FindGeographicLib.cmake  is: 
> include (FindPackageHandleStandardArgs)
find_package_handle_standard_args (GeographicLib DEFAULT_MSG
  GeographicLib_LIBRARY_DIRS GeographicLib_LIBRARIES GeographicLib_INCLUDE_DIRS)
mark_as_advanced (GeographicLib_LIBRARY_DIRS GeographicLib_LIBRARIES
  GeographicLib_INCLUDE_DIRS)

Should i unintall the apt-get installed lib, and then install from source as  [here](https://geographiclib.sourceforge.io/html/install.html)? Thanks a lot !

narutojxl, 2020-09-24:
After uninstall `sudo apt-get remove --purge geographiclib-doc    geographiclib-tools `, then build from source geographiclib-r1.50.1, install in the default `/usr/local`, and modify CMakeLists.txt of robot_localization 
> 
#set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} "/usr/share/cmake/geographiclib/")
set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} "/usr/local/share/cmake/GeographicLib/") #jxl
`catkin_make` OK.