cra-ros-pkg/robot_localization #593: Diagnostic frequency monitor min/max values are 0.

tgreier, 2020-09-15:
https://github.com/cra-ros-pkg/robot_localization/blob/285a38614135f064ced8f2532530fca271eacd87/src/ros_filter.cpp#L1956

The min/max values input to the frequency monitor must be on heap (not stack). Pointers to those memory locations are used by the diagnostic monitor task and therefore can end up accessing stack memory (reads) when reading the min/max frequency thresholds.

ayrton04, 2020-09-15:
Woof, that's been around since the beginning. Thanks for catching it. I'd welcome a PR to fix it!

ayrton04, 2020-09-18:
Oh! I just realized this was for ROS2. The ROS1 version uses [member variables](https://github.com/cra-ros-pkg/robot_localization/blob/melodic-devel/src/ros_filter.cpp#L133).

tgreier, 2020-09-18:
Ah, so this just got broke in ROS2? Is there a reason why they were changed to local variables in ROS2?
Any idea why those tests are failing in #594?

PR #594 (2020-10-02): Fix bug in creating frequency diag.
Resolves #593 
Moved the min and max frequency values to the heap since pointers to those values are passed to diagnostic_updater::FrequencyStatusParam (to allow the values to be updated on the fly).