cra-ros-pkg/robot_localization #603: Fusing the  gps and imu datas, the z coordinate data he

keguo1119, 2020-10-14:
## Bug report

**Required Info:**

- Operating System:
  -  OS and version (e.g. Ubuntu 18.04) 
- Installation type:
  -  from source 
- Version or commit hash:
  -  melodic-devel, HEAD, 9277d982fa1a2d2a05eec4f59b4f8696ca903f1c 

#### Steps to reproduce issue
I used the test1.bag to test the pkg performance and the z coordinate data has drifted as the follow pictures.
Can anybody explain why?
[image]
[image]

ayrton04, 2020-10-14:
Hi, all questions should be asked on answers.ros.org. Please post the config you are using.

closed 2020-10-14 by ayrton04

keguo1119, 2020-10-14:
[image]
[image]

ayrton04, 2020-10-14:
Please post this questions on answers.ros.org.