cra-ros-pkg/robot_localization #606: Using autograd for internal motion and measurement model of ekf_localization_node

plafer, 2020-10-24:
## Feature request
We could have the users specify the motion and measurement models, and using some autograd library, automatically generate the jacobians.

#### Feature description

The current problems I see with the library is:
1. it uses a one-size-fits-all  motion model
2. it assumes that the measurement function updates state variables directly

Having the users specify the models would (most likely):
1. improve the motion model as users can curate the equation to their platform
2. make the kalman gain "more accurate" in that it incorporates information as to how the state generates the measurements with the measurement jacobian.

We could also leave the current motion & measurement models by default for convenience.

#### Implementation considerations

I was thinking we could use some autograd solution to compute the jacobians needed in the EKF, as opposed to having the user specify both the models + jacobians, which is error-prone. I am familiar with pytorch's autograd, but maybe there are better options out there.

I volunteer to implement these changes. However, as this would probably require the move to version 3.0, I would like to discuss the changes with the community first.

ayrton04, 2020-10-28:
Thanks for your thoughts. The problem with specifying other models is that you'll likely have to do more than just implement a new filter class. Without looking at it too deeply, I think a lot of the measurement preparation steps would need to also be adjusted for each model.

I struggle with whether we want to implement this here, or encourage people to move to fuse:

https://github.com/locusrobotics/fuse

It is much more suited to writing plugins for different kinematic models, and is generally a much more flexible package. It just needs more documentation, and doesn't yet support ROS2. But I'd be interested in hearing other opinions. If we're going this far with this package, a lot of other code could stand to be refactored, like the extremely repetitive parameter loading we do.

Timple, 2020-10-28:
I think such a transition would be boosted if not only more documentation was provided but a migration guide from robot_localization. 

(I remember from the past that fuse only does 2D? But I cannot find that reference anymore).

ayrton04, 2020-10-28:
It currently only has 2D models, true:

https://github.com/locusrobotics/fuse/tree/devel/fuse_models/src

...but adding 3D ones is pretty straightforward. But yes, agreed that it seems like an important feature and the documentation is needed. In any case, this post was about adding new kinematic models, and as long as we're adding new models, it seemed easier to add them to a package that supports it.

I need to find time to work on a migration guide.

SteveMacenski, 2021-05-20:
closing - this is unlikely to ever be merged into this project as its late in its lifecycle. If you submit a PR, we will of course consider it, but this isn't a realistic request for maintainers / other users at this point

closed 2021-05-20 by SteveMacenski