cra-ros-pkg/robot_localization #607: Eloquent-devel compilation issues

DronistB, 2020-10-25:
## Bug report

**Required Info:**

- Operating System:
 18.04
- Installation type:
eloquent - debian
robot localization -from source
- Version or commit hash:
 eloquent-devel

#### Steps to reproduce issue
colcon build

#### Expected behavior
no compiling issues
#### Actual behavior
                                                                
-```
-- stderr: robot_localization                                
In file included from /opt/ros/eloquent/include/geometry_msgs/msg/transform_stamped__struct.hpp:8:0,
                 from /opt/ros/eloquent/include/geometry_msgs/msg/transform_stamped.hpp:7,
                 from ~/ros2/src/robot_localization/src/ros_filter_utilities.cpp:33:
/opt/ros/eloquent/include/rosidl_generator_cpp/bounded_vector.hpp:480:32: error: ‘emplace_back’ function uses ‘auto’ type specifier without trailing return type
   emplace_back(Args && ... args)
                                ^
/opt/ros/eloquent/include/rosidl_generator_cpp/bounded_vector.hpp:480:32: note: deduced return type only available with -std=c++14 or -std=gnu++14
In file included from /opt/ros/eloquent/include/builtin_interfaces/msg/duration__struct.hpp:8:0,
                 from /opt/ros/eloquent/include/builtin_interfaces/msg/duration.hpp:7,
                 from /opt/ros/eloquent/include/rclcpp/duration.hpp:20,
                 from ~/ros2/src/robot_localization/include/robot_localization/filter_utilities.hpp:37,
                 from ~/ros2/src/robot_localization/include/robot_localization/filter_base.hpp:38,
                 from ~/ros2/src/robot_localization/src/filter_base.cpp:34:
/opt/ros/eloquent/include/rosidl_generator_cpp/bounded_vector.hpp:480:32: error: ‘emplace_back’ function uses ‘auto’ type specifier without trailing return type
   emplace_back(Args && ... args)
                                ^
/opt/ros/eloquent/include/rosidl_generator_cpp/bounded_vector.hpp:480:32: note: deduced return type only available with -std=c++14 or -std=gnu++14
In file included from /opt/ros/eloquent/include/builtin_interfaces/msg/duration__struct.hpp:8:0,
                 from /opt/ros/eloquent/include/builtin_interfaces/msg/duration.hpp:7,
                 from /opt/ros/eloquent/include/rclcpp/duration.hpp:20,
                 from /opt/ros/eloquent/include/rclcpp/qos.hpp:21,
                 from ~/ros2/src/robot_localization/src/ros_robot_localization_listener.cpp:33:
/opt/ros/eloquent/include/rosidl_generator_cpp/bounded_vector.hpp:480:32: error: ‘emplace_back’ function uses ‘auto’ type specifier without trailing return type
   emplace_back(Args && ... args)
                                ^
/opt/ros/eloquent/include/rosidl_generator_cpp/bounded_vector.hpp:480:32: note: deduced return type only available with -std=c++14 or -std=gnu++14
In file included from /opt/ros/eloquent/include/builtin_interfaces/msg/duration__struct.hpp:8:0,
                 from /opt/ros/eloquent/include/builtin_interfaces/msg/duration.hpp:7,
                 from /opt/ros/eloquent/include/rclcpp/duration.hpp:20,
                 from ~/ros2/src/robot_localization/include/robot_localization/filter_utilities.hpp:37,
                 from ~/ros2/src/robot_localization/src/navsat_transform.cpp:34:
/opt/ros/eloquent/include/rosidl_generator_cpp/bounded_vector.hpp:480:32: error: ‘emplace_back’ function uses ‘auto’ type specifier without trailing return type
   emplace_back(Args && ... args)
                                ^
/opt/ros/eloquent/include/rosidl_generator_cpp/bounded_vector.hpp:480:32: note: deduced return type only available with -std=c++14 or -std=gnu++14
In file included from /opt/ros/eloquent/include/builtin_interfaces/msg/duration__struct.hpp:8:0,
                 from /opt/ros/eloquent/include/builtin_interfaces/msg/duration.hpp:7,
                 from /opt/ros/eloquent/include/rclcpp/duration.hpp:20,
                 from ~/ros2/src/robot_localization/include/robot_localization/filter_utilities.hpp:37,
                 from ~/ros2/src/robot_localization/include/robot_localization/filter_base.hpp:38,
                 from ~/ros2/src/robot_localization/include/robot_localization/ekf.hpp:37,
                 from ~/ros2/src/robot_localization/src/ekf.cpp:34:
/opt/ros/eloquent/include/rosidl_generator_cpp/bounded_vector.hpp:480:32: error: ‘emplace_back’ function uses ‘auto’ type specifier without trailing return type
   emplace_back(Args && ... args)
                                ^
/opt/ros/eloquent/include/rosidl_generator_cpp/bounded_vector.hpp:480:32: note: deduced return type only available with -std=c++14 or -std=gnu++14
In file included from /opt/ros/eloquent/include/builtin_interfaces/msg/duration__struct.hpp:8:0,
                 from /opt/ros/eloquent/include/builtin_interfaces/msg/duration.hpp:7,
                 from /opt/ros/eloquent/include/rclcpp/duration.hpp:20,
[... 1245 lines cut ...]
/opt/ros/eloquent/include/rclcpp/node_impl.hpp:205:53: error: use of ‘auto’ in lambda parameter declaration only available with -std=c++14 or -std=gnu++14
     [this, &normalized_namespace, ignore_overrides](auto element) {
                                                     ^~~~
/opt/ros/eloquent/include/rclcpp/node_impl.hpp: In lambda function:
/opt/ros/eloquent/include/rclcpp/node_impl.hpp:208:42: error: request for member ‘first’ in ‘element’, which is of non-class type ‘int’
           normalized_namespace + element.first,
                                          ^~~~~
/opt/ros/eloquent/include/rclcpp/node_impl.hpp:209:19: error: request for member ‘second’ in ‘element’, which is of non-class type ‘int’
           element.second.first,
                   ^~~~~~
/opt/ros/eloquent/include/rclcpp/node_impl.hpp:210:19: error: request for member ‘second’ in ‘element’, which is of non-class type ‘int’
           element.second.second,
                   ^~~~~~
/opt/ros/eloquent/include/rclcpp/node_impl.hpp: In member function ‘bool rclcpp::Node::get_parameter(const string&, ParameterT&) const’:
/opt/ros/eloquent/include/rclcpp/node_impl.hpp:228:59: error: ‘class rclcpp::Parameter’ has no member named ‘get_value’; did you mean ‘get_name’?
     parameter = static_cast<ParameterT>(parameter_variant.get_value<ParameterT>());
                                                           ^~~~~~~~~
                                                           get_name
/opt/ros/eloquent/include/rclcpp/node_impl.hpp:228:79: error: expected primary-expression before ‘>’ token
     parameter = static_cast<ParameterT>(parameter_variant.get_value<ParameterT>());
                                                                               ^
/opt/ros/eloquent/include/rclcpp/node_impl.hpp:228:81: error: expected primary-expression before ‘)’ token
     parameter = static_cast<ParameterT>(parameter_variant.get_value<ParameterT>());

```

SteveMacenski, 2020-10-26:
Are you not compiling in C++14?

DronistB, 2020-10-27:
> Are you not compiling in C++14?

I've printed the CMAKE_CXX_FLAGS in the main CMakeLists.txt
by : message(${CMAKE_CXX_FLAGS})

And it printed:
-std=c++14

SteveMacenski, 2020-10-27:
Ok, I no longer have a 18.04 machine to help debug, but if you find the root cause, we'd love a PR to resolve your issues!

SteveMacenski, 2020-10-27:
Keep in mind Eloquent goes out of support in about 4 days https://index.ros.org/doc/ros2/Releases/ so you won't receive any support from anyone from it after that point.

DronistB, 2020-10-28:
> Keep in mind Eloquent goes out of support in about 4 days https://index.ros.org/doc/ros2/Releases/ so you won't receive any support from anyone from it after that point.

Thanks @SteveMacenski,
BTW, would you recommend moving back to Dashing due to that issue? Although eloquent is a newer version?
Given a situation where I have the option to work only with 18.04 because I work with Jetson TX2.
Thanks

SteveMacenski, 2020-10-28:
I think a newer jetpack is 20.04, but if I’m mistaken, yes Dashing would work. 

Though remember docker exists....

DronistB, 2020-10-28:
> I think a newer jetpack is 20.04, but if I’m mistaken, yes Dashing would work.
> 
> Though remember docker exists....

According to Nvidia;
It’s under planning, but no clear schedule yet, at least it won’t happen before 2021 H1.
https://developer.nvidia.com/embedded/develop/roadmap

miku54, 2021-04-07:
Hello, I have the same problem as you. Have you solved it?

technixol, 2021-05-01:
I also have the same issue. Was this ever resolved?

technixol, 2021-05-01:
I solved the issue by adding these lines in my cmake:

set(CMAKE_CXX_STANDARD 14)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
set(CMAKE_CXX_EXTENSIONS OFF)

Even though my CMake too printed "-std=c++14" as the message output, I still had issues until I added these lines. This may not be an efficient solution for everyone, but it caused the package to successfully build so maybe it helps considering the EOL for eloquent.

SteveMacenski, 2021-05-20:
Closing - eloquent is EOL

closed 2021-05-20 by SteveMacenski

Exboom, 2022-04-11:
If you stumbled upon this in search of a solution to the problem, then try using the `--cmake-args -DCMAKE_CXX_STANDARD=14` flag. I don't know why, but this is what helped me

miku54, 2022-04-12:
Thank you! I solved it this way too, but I didn't update the follow-up