cra-ros-pkg/robot_localization #609: UTM to map transform not being broadcasted anymore after GPS signal is lost

ignaciotb, 2020-11-12:
Hi, 
I'm not sure this is a bug, but we're working with an AUV and once we've got a GPS fix and the UTM to map transform, we dive and we'd like that transform to still be broadcasted during the dive. This doesn't seem to be the case. Is there a reason for it? And if so, is there a workaround to get this behavior on our setup?

**Required Info:**
Ubuntu 18.04
Binaries

ayrton04, 2020-11-12:
Hi,

Thanks for the question. However, all questions should be asked on answers.ros.org.

closed 2020-11-12 by ayrton04

ignaciotb, 2020-11-12:
You're right, thanks. I've posted the question here now
https://answers.ros.org/question/365405/utm-to-map-transform-not-being-broadcasted-anymore-after-gps-signal-is-lost/