cra-ros-pkg/robot_localization #610: The problem when I try to fusing uwb & odom & imu

wangshanpao1991, 2020-11-13:
## Feature request
I want someone can help me to answer why the initial state is very import when fusing different sensor message?
And how to realize that fusing sensor at any position in map?
#### Feature description
1.  In the application with UWB, the world frame is map frame provided by the anchors of UWB sets.
2.  The robot stands at a random position in the map when I begin the fusing process by runing the robot_localization node.
    If  the random position is unknown, the odom and base_link will appear at the orgin point of map frame. And the result of 
   fusing will diverge with a inconceivable transfromation between odom frame and map frame.
3. After  setting  a initial state by putting the robot at a specified position with a fixed pose,  the robot_localization node can work but can't for a long time. Meanwhile, the odom frame will deviate from the origin position in rviz after moving according to a closed  square。 In other words, the robot is back to the initial position in real world, but only back to the origin point of odom frame which exists a deviation beteween the initial time and final time.
4. I don't know how to cope with it. Because I want to achieve a application that the robot begin to fusing at a unknown position and can work lasting.

ayrton04, 2020-11-13:
This seems a bit like a question, rather than a feature request. Can you ask it on answers.ros.org? If we can't figure out a solution, we can re-open this.

closed 2020-11-13 by ayrton04

wangshanpao1991, 2020-11-13:
Ok, thank you. By the way, could you give me some suggestion for this question?

ayrton04, 2020-11-13:
I'm afraid I don't have the cycles to look at it right now, but I'll try to get over to ROS Answers again soon.