cra-ros-pkg/robot_localization #619: Robot_localization services /toLL and /fromLL respond with different result on the same data

kosmastsk, 2021-01-29:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Binaries
- Version or commit hash:
  - 2.6.9

#### Steps to reproduce issue
I am using the robot_localization package and in the terminal, I run the services `/fromLL` and `/toL`L for the same position. But the results do not agree. Specifically, running:  
`rosservice call /robot/fromLL "ll_point:
  latitude: 40.5665059791
  longitude: 22.9989873398
  altitude: 122.0"`
returns:
`map_point: 
  x: 140892.72962
  y: 5405179.36945
  z: 0.0`
And when I run:
`rosservice call /robot/toLL "map_point:
  x: 0.0
  y: 0.0
  z: 0.0"`
I get:
`ll_point: 
  latitude: 40.5665059791
  longitude: 22.9989873398
  altitude: 122.375866815`
which is the same position as the first one. 

#### Expected behavior
The lat/long and the map points should be the same in both cases, no matter the direction that the conversion is asked.

#### Additional information
The solution is described in ros.answers here: [https://answers.ros.org/question/369570/robot_localization-services-toll-and-fromll-respond-with-different-result/](https://answers.ros.org/question/369570/robot_localization-services-toll-and-fromll-respond-with-different-result/)

ayrton04, 2021-01-29:
Thanks for the report. Will fix this ASAP.

ayrton04, 2021-02-03:
Fixed with https://github.com/cra-ros-pkg/robot_localization/pull/620. This doesn't appear to affect the foxy ROS2 branch, but the fix is now in melodic and noetic.

closed 2021-02-03 by ayrton04