cra-ros-pkg/robot_localization #625: Release strategy

Timple, 2021-02-23:
I wanted to work on #222 but noticed the default branch is `melodic-devel`. But there is also a `noetic-devel` branch. 

The bloom releases for melodic and noetic are also different.

Why the difference? 
What ROS1 release should I target? 
And will this feature be back/forward ported to the other release?

ayrton04, 2021-02-23:
As with many ROS packages, once a release is out, we really should shift development focus to bug fixes (though it obviously doesn't always go that way), and new features should go into the next release. This package has a branch for every ROS release since ROS Groovy, so we're carrying on with the same process that we always have.

But more importantly, there were changes to `tf2` in Noetic that necessitated a Noetic branch.

Anyway, my recommendation would be to target `noetic-devel`, and if the change is lightweight enough, I'll consider cherry picking into `melodic-devel`.

closed 2021-02-23 by ayrton04

Timple, 2021-02-23:
Would it make sense to make noetic-devel the default branch in that case? Because that is what threw me off in the first place.

ayrton04, 2021-02-23:
Sure, I can do that.