cra-ros-pkg/robot_localization #630: pose_rejection_threshold documention unclear

jola6897, 2021-03-04:
## Add more details to the documentation of pose_rejection_threshold calibration parameters.
If I understand the code correctly  *_pose_rejection_threshold squared shall be the chi-square distribution quantile for a given degree of freedom in a chosen confidence.

#### Feature description
Given an input with a degree of freedom of 1 e.g. measured x-position, what confidence are you choosing? 0.05 [(chi-square quantiles table)](https://en.wikipedia.org/wiki/Chi-square_distribution)? With 1 DoF and 0.05 this would mean this is 3.84 which results in *_pose_rejection_threshold to be 1.96 approximately. 

Could you maybe elaborate on one example from the [template](https://github.com/cra-ros-pkg/robot_localization/blob/noetic-devel/params/ukf_template.yaml) how you came up with the proposed values? Since I also do not see a test using the a rejection threshold that is not the numeric maximum.

I would be very glad if you could shed some light into this topic. Best regards.

ayrton04, 2021-03-09:
The rejection threshold is a [Mahlanobis distance](https://github.com/cra-ros-pkg/robot_localization/blob/a2da33bb3aaa4b10358de391221fe0ed97b16f49/params/ekf_template.yaml#L117). The original PR is [here](https://github.com/cra-ros-pkg/robot_localization/pull/146).

In a 1D problem (we'll continue to use X in this example), it's effectively just an unsigned Z-score. We're measuring how many standard deviations from the mean a given measurement is:

```
distance = sqrt( (x_measured - x_state) / x_variance )
```

If distance > threshold, we reject the measurement (don't fuse it). This was just a way to reject outlier measurements before they had the chance to destroy the state estimate.

The values in the template are just examples. The template doesn't refer to any specific state estimation implementation, so the values are arbitrary.

jola6897, 2021-03-09:
Thanks you a lot for the explanation.

closed 2021-03-09 by jola6897

nyxrobotics, 2022-09-29:
I want to adjust the parameters, is there a way to publish the current Mahalanobis distance as debugging information?

nyxrobotics, 2022-10-27:
This issue has not been resolved.
Can you reopen?

ayrton04, 2022-10-27:
I'll reopen it if someone would be willing to PR the fix. :)