cra-ros-pkg/robot_localization #636: Not providing "FindGeographicLib.cmake" in CMAKE_MODULE_PATH

AxelCORNILLEAU, 2021-03-18:
## Bug report

After installing GeographicLib1.51 on Jetson Xavier, CMakelist.txt points /usr/share/cmake/GeographicLib/ at set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} "/usr/share/cmake/GeographicLib/") while the installation of GeographicLib1.51 points /usr/local/share/cmake/GeographicLib/.
Replacing set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} "/usr/share/cmake/GeographicLib/") for set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} "/usr/local/share/cmake/GeographicLib/") fixed the issue

- Operating System:
  - 
- Installation type:
  - 
- Version or commit hash:
  - 

#### Steps to reproduce issue

On Jetson Xavier with Ubuntu18.04
Download and install GeographicLib1.51 with https://geographiclib.sourceforge.io/html/install.html
Download robot_localisation from git
#### Expected behavior
set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} "/usr/share/cmake/GeographicLib/")
#### Actual behavior
set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} "/usr/local/share/cmake/GeographicLib/")
#### Additional information
reported also [Here](https://answers.ros.org/question/374191/geographiclib151-on-jetson-xavier-for-robot_localization/)

SteveMacenski, 2021-03-18:
Did you install GeographicLib or use rosdep? 

Also, `latest` is not a valid version, please fill out the template with your actual info.

AxelCORNILLEAU, 2021-03-18:
I did not. I mostly raised this issue for people who will google the issue, as this is how most of people do. Also thx for the rosdep tip, I didn't used it.

SteveMacenski, 2021-03-18:
OK, closing - ROS Answers was probably the right place for this problem

closed 2021-03-18 by SteveMacenski