cra-ros-pkg/robot_localization #644: base_link with namespaces issue. "Could not obtain transform from base_link"

rakshith95, 2021-04-05:
Hello, 

```
 <group if="$(arg enable_ekf)" >   
   <node pkg="robot_localization" type="ekf_localization_node" name="ekf_localization">    
      <rosparam command="load" file="$(arg ekf_yaml_path)" />  
      <param name="map_frame"       value="map"/>  
      <param name="odom_frame"      value="$(arg ns)/odom"/> 
      <param name="base_link_frame" value="$(arg ns)/base_link"/> 
      <param name="world_frame"     value="$(arg ns)/odom"/>  
    </node>  
  </group>  
```

The above are the relevant lines in my launch file, but when I try it to run it, there are warnings thrown and the functionality is not as expected. The warnings are as below:

```
[WARN] [1617652671.519795535, 1553.900000000]: Could not obtain transform from base_link to jackal2/base_link. Error was "base_link" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1617652673.364580966, 1555.740000000]: Could not obtain transform from base_link to jackal1/base_link. Error was "base_link" passed to lookupTransform argument source_frame does not exist. 
```
I don't understand why these would come up since I am specifically assigning a base_link_frame. Why does it matter if "base_link" exists or not?

SteveMacenski, 2021-04-05:
Please look up answers to this question on ROS Answers, I'm sure you'll find it - else please ask a new question there with the pertinent information.

closed 2021-04-05 by SteveMacenski

rakshith95, 2021-04-06:
I have asked there , but I thought it would be pertinent here as well since the behaviour isn't expected.

SystemDiagnosticss, 2021-05-12:
@rakshith95 Are you solved?

rakshith95, 2021-05-12:
@SystemDiagnosticss yeah I was able to solve this. I got the answer from my [post](https://answers.ros.org/question/375480/warnings-with-base_link-in-namespace-for-robot_localization/) on ROS answers.