cra-ros-pkg/robot_localization #651: using world frame as earth

TZVIh, 2021-04-14:
Hi, I am using robot localization for my robot.

I have two filters, one for Odom->baselink and one for map->odom

The map topic is from a SLAM node.

In my system, I have a world frame that represents a shared map.

I have a static tf publisher for world->map

I want that the output of the filter will be on the world frame.

I changed the configuration to be like this:

            map_frame: map

            odom_frame: odom

            base_link_frame: base_link

            world_frame: world

I got "Odometry message frame_id was world, expected map or doom" error.

How can I get the output of the filter on the world (earth) frame?

Thanks.

ayrton04, 2021-04-14:
Thanks for the question. However, I ask that all questions be asked on answers.ros.org. If you have already asked there, then I will answer as soon as I can.

closed 2021-04-14 by ayrton04

TZVIh, 2021-04-14:
I am sorry for my misunderstanding.
I opened on answer.ros.org
https://answers.ros.org/question/376102/robot-localization-using-world-frame/

Thanks!