cra-ros-pkg/robot_localization #656: navsat_transform_node /fromLL service incorrect output

r0gi, 2021-04-29:
## Bug report

**Required Info:**

Operating System: Ubuntu 20.04.2 LTS
Apt Package: ros-noetic-robot-localization
Version: 2.7.1-1focal.20210126.131257
APT-Sources: http://packages.ros.org/ros/ubuntu focal/main amd64 Packages

#### Steps to reproduce issue
Launch navsat_transform_node:
```
<node pkg="robot_localization" type="navsat_transform_node" name="navsat_transform_node" respawn="true">
<param name="broadcast_cartesian_transform" value="true"/>
</node>
```
Publish to GPS topic to establish map -> utm transform: (assuming robot is publishing valid imu etc. already)
```
rostopic pub /gps/fix sensor_msgs/NavSatFix "header:
  seq: 1
  stamp: {secs: 0, nsecs: 0}
  frame_id: 'gps'
status: {status: 0, service: 0}
latitude: 42.000  
longitude: 146.000  
altitude: 1.0
position_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
position_covariance_type: 1"
```

Call `/toLL` service, which returns latitude and longitude as expected:

```
rosservice call /toLL "map_point:
  x: 0.0                         
  y: 0.0              
  z: 0.0" 
```
```
ll_point: 
  latitude: 42.00000020241304
  longitude: 145.99999953120394
  altitude: 0.3590296234368515
```

Call `/fromLL` service, which returns some strange values, expected x: 0.0 and y: 0.0
```
rosservice call /fromLL "ll_point:
  latitude: 42.000
  longitude: 146.000
  altitude: 0.0" 
```
```
map_point: 
  x: 2314987.444692288
  y: -5520754.56380302
  z: -0.3590296234368515
```

Timple, 2021-04-29:
I've rewritten a lot of these conversions last week. They are not in the binaries yet.
Would you mind checking if the problem reproduces if you build from source? (`noetic-devel` branch)

r0gi, 2021-04-29:
I actually made the mistake of cloning the repo and going over the source code, but using the apt release to test the service calls.

Testing the package after building from source:
```
rosservice call /fromLL "ll_point:
  latitude: 42.000
  longitude: 146.000
  altitude: 0.0"
``` 
```
map_point: 
  x: 0.024438641034066677
  y: -0.03800516342744231
  z: -0.3590296234368515
```
I assume the error comes from the UTM transform itself within the geographiclib and can't be improved. In any case it's good enough for my use case. Thank you!

closed 2021-04-29 by r0gi