cra-ros-pkg/robot_localization #660: ROS2 Foxy Warning: Invalid frame ID "imu" passed to canTransform argument source frame ~ frame does not exist

sisaha9, 2021-05-10:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - Binaries ROS2 Foxy
- Version or commit hash:
  - 

#### Steps to reproduce issue
Use the dual ekf configuration here: https://github.com/cra-ros-pkg/robot_localization/blob/foxy-devel/launch/dual_ekf_navsat_example.launch.py . But with all the imu/data replaced with imu as per #658 and the other raw sensor information changed to the ones I use

Here is the transform tree published. The base_link and children being generated by a URDF and robot_state_publisher in ROS2. The map -> odom and odom -> base_link should be coming from the dual ekf configuration linked above

[image]

Here is the YAML file:
```yaml
ekf_local_filter_node:
    ros__parameters:
        use_sim_time: true
        publish_tf: true
        frequency: 30.0
        sensor_timeout: 0.1
        two_d_mode: false
        transform_time_offset: 0.0
        transform_timeout: 0.0
        print_diagnostics: true
        debug: false

        map_frame: map
        odom_frame: odom
        base_link_frame: base_link
        world_frame: odom

        odom0: /lgsvl/gnss_odom
        odom0_config: [false, false, false,
                      false, false, false,
                      true,  true,  true,
                      false, false, true,
                      false, false, false]
        odom0_queue_size: 10
        odom0_nodelay: true
        odom0_differential: false
        odom0_relative: false

        imu0: /imu/imu_raw
        imu0_config: [false, false, false,
                      true,  true,  false,
                      false, false, false,
                      true,  true,  true,
                      true,  true,  true]
        imu0_nodelay: false
        imu0_differential: false
        imu0_relative: false
        imu0_queue_size: 10
        imu0_remove_gravitational_acceleration: true

        use_control: false

        process_noise_covariance: [1e-3, 0.0,    0.0,    0.0,    0.0,    0.0,    0.0,     0.0,     0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,
                                  0.0,    1e-3,  0.0,    0.0,    0.0,    0.0,    0.0,     0.0,     0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,
                                  0.0,    0.0,    1e-3,  0.0,    0.0,    0.0,    0.0,     0.0,     0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,
                                  0.0,    0.0,    0.0,    0.3,   0.0,    0.0,     0.0,     0.0,     0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,
                                  0.0,    0.0,    0.0,    0.0,    0.3,   0.0,    0.0,     0.0,     0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,
                                  0.0,    0.0,    0.0,    0.0,    0.0,    0.01,  0.0,     0.0,     0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,
                                  0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.5,    0.0,     0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,
                                  0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,     0.5,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,
[... 178 lines cut ...]
[ekf_node-2]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[ekf_node-2] Warning: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist
[ekf_node-2]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[ekf_node-2] Warning: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist
[ekf_node-2]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[ekf_node-2] Warning: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist
[ekf_node-2]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[ekf_node-2] Warning: Invalid frame ID "imu" passed to canTransform argument source_frame - frame does not exist
[ekf_node-2]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[ekf_node-2] Warning: Invalid frame ID "imu" passed to canTransform argument source_frame - frame does not exist
[ekf_node-2]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[ekf_node-2] Warning: Invalid frame ID "imu" passed to canTransform argument source_frame - frame does not exist
[ekf_node-2]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[navsat_transform_node-3] [INFO] [1620608307.556481401] [navsat_transform_node]: Datum (latitude, longitude, altitude) is (37.42, -122.02, -0.72)
[navsat_transform_node-3] [INFO] [1620608307.556658539] [navsat_transform_node]: Datum UTM coordinate is (10S, 587061.32, 4141628.66)
[navsat_transform_node-3] [INFO] [1620608307.586789412] [navsat_transform_node]: Corrected for magnetic declination of 0, user-specified offset of 0 and meridian convergence of 0.0104345. Transform heading factor is now -1.81913
```

Sometimes it happens in ekf_node-1 and sometimes in navsat_transform_node. Sometimes I will get the following additional message after the "Corrected for magnetic declination of 0"
```bash
[navsat_transform_node-3] [ERROR] [1620612713.623615821] [navsat_transform_node]: Could not obtain map -> base_link transform. Will not remove offset of navsat device from robot's origin
```

#### Additional information
All data is produced by the SVL simulator. I can verify that the data is right though

ayrton04, 2021-05-10:
If you look where the warnings are coming from:

```
at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
```

That's happening at the level of `tf2` itself. It's complaining that it doesn't know about those frames. The fact that the warning goes away means those transforms eventually _are_ present. This suggests to me some kind of lifecycle issue. Are you sure the robot state publisher is completely launched _before_ `navsat_transform_node`? And has the `odom`-frame EKF already started up and published at least one transform? I don't use ROS2, but I believe it supports lifecycle management. I would double-check that your nodes are starting up in order:

- robot state publisher
- tier 1 EKF
- tier 2 EKF
- navsat_transform_node

sisaha9, 2021-05-11:
I see. That is likely the issue. Will try that. Thanks!

srujan-aris, 2021-07-06:
@sisaha9, were you able to find out what the issue was about? I get the same error, however for a different usecase though

sisaha9, 2021-07-06:
I was doing this on the SVL simulator and I guess it lagged my machine enough for transforms and messages to publish late

srujan-aris, 2021-07-06:
Thanks for your response. I was able to resolve this issue by explicitly starting joint_state_publisher node through the launch file, along with my main node. That kind of fixed my issue. More information on this other discussion [here](https://github.com/ros-controls/ros2_control_demos/issues/107)

OguzKahramn, 2022-06-28:
@srujan-aris Hi. I have the same problem even tough I am starting joint_state_publisher node inside the launch file. I have not found a solution through the link you sent as well. Could you help me please?

ayrton04, 2022-06-29:
Please ask questions on answers.ros.org or robotics.stackexchange.com.

bartboogmans, 2023-11-06:
Why is it required that this frame ID needs to be previously known to avoid this error?