cra-ros-pkg/robot_localization #665: the tf2 dependency will be automatically indexed to the melodic directory

miku54, 2021-05-21:
## Bug report

**Required Info:**

- Operating System:
  - ubuntu18.04
- Installation type:
  - source
- Version or commit hash:
  - eloquent

#### Steps to reproduce issue
colcon build --packages-select robot_localization
#### Expected behavior
Summary: 1 package finished
#### Actual behavior
/opt/ros/melodic/include/tf2_ros/transform_broadcaster.h:51:3: note:   candidate expects 0 arguments, 1 provided
/opt/ros/melodic/include/tf2_ros/transform_broadcaster.h:48:7: note: candidate: tf2_ros::TransformBroadcaster::TransformBroadcaster(const tf2_ros::TransformBroadcaster&)
 class TransformBroadcaster{
       ^~~~~~~~~~~~~~~~~~~~
/opt/ros/melodic/include/tf2_ros/transform_broadcaster.h:48:7: note:   no known conversion for argument 1 from ‘std::shared_ptr<rclcpp::Node>’ to ‘const tf2_ros::TransformBroadcaster&’
/opt/ros/melodic/include/tf2_ros/transform_broadcaster.h:48:7: note: candidate: tf2_ros::TransformBroadcaster::TransformBroadcaster(tf2_ros::TransformBroadcaster&&)
/opt/ros/melodic/include/tf2_ros/transform_broadcaster.h:48:7: note:   no known conversion for argument 1 from ‘std::shared_ptr<rclcpp::Node>’ to ‘tf2_ros::TransformBroadcaster&&’
make[2]: *** [CMakeFiles/rl_lib.dir/src/ros_filter.cpp.o] Error 1
make[1]: *** [CMakeFiles/rl_lib.dir/all] Error 2
make: *** [all] Error 2
#### Additional information
In the build/robot_localization/CmakeCache.txt file, the directories are all the directories of the melodic system

SteveMacenski, 2021-05-21:
This is not a robot_localization issue, please ask on ROS Answers - you're mixing paths for ROS2 and ROS1, you cannot do that, keep them isolated

miku54, 2021-05-21:
Okay, thank you.

closed 2021-06-03 by ayrton04