cra-ros-pkg/robot_localization #680: Potential mathematics mistake

tchoopojcharoen, 2021-07-16:
https://github.com/cra-ros-pkg/robot_localization/blob/bab5f263e7febcca0fdf1d6ef1c348122b92a835/src/ekf.cpp#L327

I think it should be

`double dFY_dP = (-sr * cpi * cpi * pitchVel - cr * cpi * cpi * yawVel) * delta;
`
But I'm not 100% certain. This is based on the mapping between the angular velocity and the orientation that the author wrote in the code. I determine the nonlinear function and perform differentiation to find this particular part of the Jacobian.

ayrton04, 2021-07-16:
See https://github.com/cra-ros-pkg/robot_localization/issues/591 and https://github.com/cra-ros-pkg/robot_localization/pull/592.

tchoopojcharoen, 2021-07-16:
After brief rework on my math, I've realized I made a calculation mistake by not applying the proper chain rules. It matched with the code now. Thank you so much for making times for this, and sorry for wasting your time. At least, I'm able to figure out the underlying model that you've been using. :)

closed 2021-07-16 by tchoopojcharoen

ayrton04, 2021-07-16:
Happy to help! Thanks for reviewing.