cra-ros-pkg/robot_localization #681: Inconsistent Jacobian dFR/dR with 1.0

tchoopojcharoen, 2021-07-16:
https://github.com/cra-ros-pkg/robot_localization/blob/bab5f263e7febcca0fdf1d6ef1c348122b92a835/src/ekf.cpp#L283

Based on the prediction model. The predicted Roll-pitch-yaw can be expressed as matrix equation along the line of this.

[FR,FP;FY] = J(R,P,Y)*[Wx,Wy,Wz]

When I looked at dFR/dR, the code has this additional 1.0, which implies that FR = R+(something , which is not the case. I'm not whether 1.0 should be there unless I've got the wrong prediction model, which I'm pretty sure I've got it right from other parts of the code.

ayrton04, 2021-07-16:
https://github.com/cra-ros-pkg/robot_localization/blob/bab5f263e7febcca0fdf1d6ef1c348122b92a835/src/filter_base.cpp#L89

The transfer function starts out with the identity matrix (this gets set in `FilterBase`, not `Ekf`), so indeed, every variable is some function of its previous value. In the case of roll, the function is

`roll_pred = roll + v_roll * dt + v_pitch * sin(roll) * tan(pitch) * dt + v_yaw * cos(roll) * tan(p) * dt`

Differentiating with respect to roll should yield the equation for `dFR_dR` that we have.

The function is just taking the current value of `roll` and adding components of the other orientation velocities, based on our current orientation.

ayrton04, 2021-07-16:
I do want to make sure I didn't miss anything with the other two rotational derivatives, though.

tchoopojcharoen, 2021-07-16:
I totally miss that .setIdentity(). This makes total sense since it is the discretization of continuous system. They will always have 1 i n the diagonal. Since you set all of them in FilterBase, there is no need to write them out again unless you have something dFR_dR which is on the diagonal. Thank you for the prompt response.

closed 2021-07-16 by tchoopojcharoen