cra-ros-pkg/robot_localization #685: Error trying to run ekf_node on ros2 galactic, but it works on ros2 foxy

rythm07, 2021-07-21:
Hello,

Thanks in advance for your help!

I'm having a nightmare trying to figure out what's wrong with my code. I just installed ros2 galactic and I'm running the code that works on ros2 foxy. When I launch the robot_localization package it works fine on ros2 foxy, but in galactic I'm having this error:

**[ekf_node-1] terminate called after throwing an instance of 'rclcpp::ParameterTypeException' [ekf_node-1]   what():  expected [string] got [not set]**

I changed to galactic because It has a lot of improvements that are very useful for my thesis project so It would be much appreciated some help to get it to work in this distro.

Here is the screenshot of the terminal:

[image]

**This is my configuration file:**

`### ekf config file ###
ekf_filter_node:
    ros__parameters:
# The frequency, in Hz, at which the filter will output a position estimate. Note that the filter will not begin
# computation until it receives at least one message from one of theinputs. It will then run continuously at the
# frequency specified here, regardless of whether it receives more measurements. Defaults to 30 if unspecified.
        frequency: 30.0

# ekf_localization_node and ukf_localization_node both use a 3D omnidirectional motion model. If this parameter is
# set to true, no 3D information will be used in your state estimate. Use this if you are operating in a planar
# environment and want to ignore the effect of small variations in the ground plane that might otherwise be detected
# by, for example, an IMU. Defaults to false if unspecified.
        two_d_mode: false

# Whether to publish the acceleration state. Defaults to false if unspecified.
        publish_acceleration: false

# Whether to broadcast the transformation over the /tf topic. Defaultsto true if unspecified.
        publish_tf: true

# 1. Set the map_frame, odom_frame, and base_link frames to the appropriate frame names for your system.
#     1a. If your system does not have a map_frame, just remove it, and make sure "world_frame" is set to the value of odom_frame.
# 2. If you are fusing continuous position data such as wheel encoder odometry, visual odometry, or IMU data, set "world_frame"
#    to your odom_frame value. This is the default behavior for robot_localization's state estimation nodes.
# 3. If you are fusing global absolute position data that is subject to discrete jumps (e.g., GPS or position updates from landmark
#    observations) then:
#     3a. Set your "world_frame" to your map_frame value
#     3b. MAKE SURE something else is generating the odom->base_link transform. Note that this can even be another state estimation node
#         from robot_localization! However, that instance should *not* fuse the global data.
        map_frame: map              # Defaults to "map" if unspecified
        odom_frame: odom            # Defaults to "odom" if unspecified
        base_link_frame: base_link  # Defaults to "base_link" ifunspecified
        world_frame: odom           # Defaults to the value ofodom_frame if unspecified

        odom0: /lgsvl/gps/odom
        odom0_config: [true,  true,  true,
                       false, false, false,
                       false, false, false,
                       false, false, true,
                       false, false, false]

        imu0: /lgsvl/imu
        imu0_config: [false, false, false,
                      true,  true,  true,
                      false, false, false,
                      false, false, false,
                      false, false, false]

# [ADVANCED] The process noise covariance matrix can be difficult to tune, and can vary for each application, so it is
# exposed as a configuration parameter. This matrix represents the noise we add to the total error after each
# prediction step. The better the omnidirectional motion model matches your system, the smaller these values can be.
# However, if users find that a given variable is slow to converge, one approach is to increase the
# process_noise_covariance diagonal value for the variable in question, which will cause the filter's predicted error
# to be larger, which will cause the filter to trust the incoming measurement more during correction. The values are
[... 40 lines cut ...]
`from launch import LaunchDescription
from launch_ros.actions import Node
from ament_index_python.packages import get_package_share_directory
import os

robot_localization_params = os.path.join(
            get_package_share_directory('navigation'),
            'config',
            'robot_localization_params.yaml'
        ),

def generate_launch_description():
    return LaunchDescription([
        Node(
          package='robot_localization',
          executable='ekf_node',
          name='ekf_filter_node',
          output='screen',
          parameters=[
            robot_localization_params
            ]
        )
    ])`

Any ideas of what it can be?

rythm07, 2021-07-21:
@MCFurry

SteveMacenski, 2021-07-22:
Please search for that error, I'm sure you'll find immediately if you do so.

rythm07, 2021-07-22:
Yeah, but what I found is what the error message says. Somewhere in my parameters file, there is one parameter that should be a string, but when I replaced that file with the ekf.yaml example in this repository, it had no effect. As I am a beginner in ROS2 it could be that my launch file is not correct or so.

What is weird is that it works on foxy but does not on galactic and for what I know robot_localization didn't change for that much from foxy to galactic, at least in terms of parameters and syntax.

And that's why I posted it here because someone more experienced could spot my error.

emilnovak, 2021-07-27:
I had a similar complication yesterday. You will need to add the package to your workspace and build it from source. Either change the src/ros_filter.cpp code to match the one on the ros2 branch or fork the foxy-devel branch just cherry pick [this](https://github.com/cra-ros-pkg/robot_localization/commit/3b2cb792abe0261aa55e6f3f4ad49d7efebcdbd6) commit from the ros2 branch. This alone may not solve the whole getting the foxy branch working problem so take a look at this [issue](https://github.com/cra-ros-pkg/robot_localization/issues/622) as well.

Serafadam, 2021-08-10:
I'm having same issues, fixed them with solution above

Jconn, 2021-08-20:
to expand on what @emilnovak said, the galactic sync currently _requires_ the params in that cherry-pick to be set, things like:
```
tf_prefix
control_config
acceleration_limits,
```
if you don't have these params set, you'll get an exception on node launch. That cherry-pick commit will give all of these parameters a default value, which will stop the exception from being thrown if the params aren't manually set by the roboticist.

cesar-lopez-mar, 2021-09-14:
I have created a branch with the right cherry-picks here: https://github.com/nobleo/robot_localization/tree/fix/galactic%2Fload_parameters

SteveMacenski, 2021-09-14:
Please submit a PR

closed 2021-09-15 by SteveMacenski