cra-ros-pkg/robot_localization #686: UTM to Map not being published

Chinj17, 2021-07-29:
Hello,

I am running ROS 2 Foxy on Ubuntu 20.04. I have followed the tutorials for dual ekf and I am trying to publish the transform between UTM-->Map--->Odom. Map-->Odom is being published properly but there is no transform between UTM and Map. Also, after remapping the topics to odometry/local and odometry/global, many times I don't get odometry/local. If I keep restarting all my launch files then maybe once in 5-6 tries do I get odometry/local also. Here is my dual_ekf.yaml file. Please let me know where I am going wrong.

`

    ekf_filter_node_odom:
      ros__parameters:
        frequency: 30.0
        sensor_timeout: 0.1
        two_d_mode: true
        transform_time_offset: 0.0
        transform_timeout: 0.0
        print_diagnostics: false
        debug: false
        publish_tf: true
        publish_acceleration: true

        map_frame: map
        odom_frame: odom
        base_link_frame: chassis
        world_frame: odom

        odom0: gazebo/odom
        odom0_config: [false, false, false,
                                 false, false, false,
                                 true,  false,  false,
                                 false, false, false,
                                 false, false, false]
        odom0_queue_size: 10
        odom0_nodelay: false
        odom0_differential: false
        odom0_relative: false

        imu0: gazebo/imu
        imu0_config: [false, false, false,
                               false,  false,  true,
                               false, false, false,
                               false,  false,  true,
                               true,  false,  false]
        imu0_nodelay: false
        imu0_differential: false
        imu0_relative: false
        imu0_queue_size: 10
        imu0_remove_gravitational_acceleration: true

        use_control: false
        stamped_control: false
        control_timeout: 0.2
        control_config: [true, false, false, false, false, true]
        acceleration_limits: [1.3, 0.0, 0.0, 0.0, 0.0, 3.4]
        deceleration_limits: [1.3, 0.0, 0.0, 0.0, 0.0, 4.5]
        acceleration_gains: [0.8, 0.0, 0.0, 0.0, 0.0, 0.9]
        deceleration_gains: [1.0, 0.0, 0.0, 0.0, 0.0, 1.0]

        process_noise_covariance: [1.0,   0.0,    0.0,    0.0,    0.0,    0.0,    0.0,     0.0,     0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,
                              0.0,    1.0,    0.0,    0.0,    0.0,    0.0,    0.0,     0.0,     0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,
                              0.0,    0.0,    1e-3,   0.0,    0.0,    0.0,    0.0,     0.0,     0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,
                              0.0,    0.0,    0.0,    0.3,    0.0,    0.0,    0.0,     0.0,     0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,
                              0.0,    0.0,    0.0,    0.0,    0.3,    0.0,    0.0,     0.0,     0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,
                              0.0,    0.0,    0.0,    0.0,    0.0,    0.01,   0.0,     0.0,     0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,
                              0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.5,     0.0,     0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,
                              0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,     0.5,     0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,
                              0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,     0.0,     0.1,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,
                              0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,     0.0,     0.0,    0.3,    0.0,    0.0,    0.0,    0.0,    0.0,
                              0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,     0.0,     0.0,    0.0,    0.3,    0.0,    0.0,    0.0,    0.0,
                              0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,     0.0,     0.0,    0.0,    0.0,    0.3,    0.0,    0.0,    0.0,
                              0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,     0.0,     0.0,    0.0,    0.0,    0.0,    0.3,    0.0,    0.0,
                              0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,     0.0,     0.0,    0.0,    0.0,    0.0,    0.0,    0.3,    0.0,
                              0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.0,     0.0,     0.0,    0.0,    0.0,    0.0,    0.0,    0.0,    0.3]
[... 159 lines cut ...]

            launch_ros.actions.Node(
                package='robot_localization',
                executable='ekf_node',
                name='ekf_filter_node_map',
                output='screen',
                parameters=[parameters_file_path, {'use_sim_time': use_sim_time}],
                remappings=[('odometry/filtered', 'odometry/global')]
            ),

            launch_ros.actions.Node(
                package='robot_localization',
                executable='navsat_transform_node',
                name='navsat_transform_node',
                output='screen',
                parameters=[parameters_file_path, {'use_sim_time': use_sim_time}],
                remappings=[('imu/data', 'imu/data'),
                                      ('gps/fix', 'gps/fix'),
                                      ('gps/filtered', 'gps/filtered'),
                                      ('odometry/gps', 'odometry/gps'),
                                      ('odometry/filtered', 'odometry/global')]
                )
         ])

`

SteveMacenski, 2021-07-29:
http://docs.ros.org/en/lunar/api/robot_localization/html/navsat_transform_node.html did you read the docs and enable the appropriate parameters?

Chinj17, 2021-07-29:
Hi, thanks for replying. I have enabled the appropriate parameters. Can you tell me if something is wrong in my yaml and launch file? because I am out of ideas and solutions for this.

ayrton04, 2021-08-10:
Is there any console output when you run? Wondering if it's not getting something that it needs.