cra-ros-pkg/robot_localization #690: prepareControl didn‘t use predictionDelta as caliculate a=dv/dt?

improve100, 2021-08-09:
void FilterBase::prepareControl(const double referenceTime, const double predictionDelta)
  {
    controlAcceleration_.setZero();

    if (useControl_)
    {
      bool timedOut = ::fabs(referenceTime - latestControlTime_) >= controlTimeout_;

      if (timedOut)
      {
        FB_DEBUG("Control timed out. Reference time was " << referenceTime << ", latest control time was " <<
          latestControlTime_ << ", control timeout was " << controlTimeout_ << "\n");
      }

      for (size_t controlInd = 0; controlInd < TWIST_SIZE; ++controlInd)
      {
        if (controlUpdateVector_[controlInd])
        {
          controlAcceleration_(controlInd) = computeControlAcceleration(state_(controlInd + POSITION_V_OFFSET),
            (timedOut ? 0.0 : latestControl_(controlInd)), accelerationLimits_[controlInd],
            accelerationGains_[controlInd], decelerationLimits_[controlInd], decelerationGains_[controlInd]);
         //maybe is
          controlAcceleration_(controlInd) /= predictionDelta;
        }
      }
    }
  }

    inline double computeControlAcceleration(const double state, const double control, const double accelerationLimit,
      const double accelerationGain, const double decelerationLimit, const double decelerationGain)
    {
      FB_DEBUG("---------- FilterBase::computeControlAcceleration ----------\n");

      const double error = control - state;
      const bool sameSign = (::fabs(error) <= ::fabs(control) + 0.01);
      const double setPoint = (sameSign ? control : 0.0);
      const bool decelerating = ::fabs(setPoint) < ::fabs(state);
      double limit = accelerationLimit;
      double gain = accelerationGain;

      if (decelerating)
      {
        limit = decelerationLimit;
        gain = decelerationGain;
      }

      const double finalAccel = std::min(std::max(gain * error, -limit), limit);

      FB_DEBUG("Control value: " << control << "\n" <<
               "State value: " << state << "\n" <<
               "Error: " << error << "\n" <<
               "Same sign: " << (sameSign ? "true" : "false") << "\n" <<
               "Set point: " << setPoint << "\n" <<
               "Decelerating: " << (decelerating ? "true" : "false") << "\n" <<
               "Limit: " << limit << "\n" <<
               "Gain: " << gain << "\n" <<
               "Final is " << finalAccel << "\n");

      return finalAccel;
    }

ayrton04, 2021-08-10:
Yes, I think you're right. The [original code](https://github.com/cra-ros-pkg/robot_localization/commit/6e83ef2f3635fc84efc2413db0fc05e294c4d07e#diff-b96ffbfcfc303ae826323830be2c943272c096bf8cb93fdb56fef4c949a19ce2R283) was multiplying the error term by a frequency, and that frequency was computed as `1.0 / predictionDelta`:

https://github.com/cra-ros-pkg/robot_localization/commit/6e83ef2f3635fc84efc2413db0fc05e294c4d07e#diff-8a83a6a31c3794ee91112354ac3dcceaa70fb760089c851e5998dcc186dbaf65R329
https://github.com/cra-ros-pkg/robot_localization/commit/6e83ef2f3635fc84efc2413db0fc05e294c4d07e#diff-b96ffbfcfc303ae826323830be2c943272c096bf8cb93fdb56fef4c949a19ce2R283

But then we capped the acceleration with a gain term instead, and lost the time delta term. However, I don't think your proposed fix will work either. The acceleration limits get applied in `computeControlAcceleration`, so we need to pass the `predictionDelta` into that method so we can compute the final acceleration value within it, and then cap it. In other words, I think [this line](https://github.com/cra-ros-pkg/robot_localization/blob/da0f9676c27db71b9e7cc76fbb227c92202ffd49/include/robot_localization/filter_base.h#L360) needs to be:

`const double finalAccel = std::min(std::max(gain * error / predictionDelta, -limit), limit);`

But in order to do that, you need to pass `predictionDelta` into `computeControlAcceleration`, preferably as the third argument.

ayrton04, 2021-08-10:
Care to submit a PR?