cra-ros-pkg/robot_localization #691: Fail loading parameters in ros galactic

cesar-lopez-mar, 2021-09-08:
## Bug report
Parameters seems not be correctly loaded. Someone had also same error and the problem lies on the way parameters are defined starting from ros galactic: https://answers.ros.org/question/382967/ekf_node-1-terminate-called-after-throwing-an-instance-of-rclcppparametertypeexception-ekf_node-1-what-expected-string-got-not-set/

https://docs.ros.org/en/galactic/Releases/Release-Galactic-Geochelone.html#parameter-types-are-now-static-by-default
**Required Info:**

- Operating System:
  - 
  Ubuntu 20.04.2 LTS
- Installation type:
  - 
  Both
- Version or commit hash:
  - 
  46e035ff4f43258bee8706b32b4fca6620158f0c
 
#### Steps to reproduce issue

ros2 launch robot_localization ekf.launch.py

#### Expected behavior
ekf node is started
#### Actual behavior
ekf node fails. Output:
[ekf_node-1] 1631091950.704027 [0]   ekf_node: using network interface wlp2s0 (udp/192.168.0.193) selected arbitrarily from: wlp2s0, zt2lrxjsew, ztmjfez5yc, docker0
[ekf_node-1] terminate called after throwing an instance of 'rclcpp::ParameterTypeException'
[ekf_node-1]   what():  expected [string] got [not set]

#### Additional information

cesar-lopez-mar, 2021-09-14:
Solution provided in https://github.com/cra-ros-pkg/robot_localization/issues/685 works.

cesar-lopez-mar, 2021-09-14:
I have created a branch with the right cherry-picks here: https://github.com/nobleo/robot_localization/tree/fix/galactic%2Fload_parameters

SteveMacenski, 2021-09-14:
Duplicate of #685 closing

But sounds like a solution is found so please submit a PR and we can merge it in and re-release it

closed 2021-09-14 by SteveMacenski