cra-ros-pkg/robot_localization #693: Cannot catkin_make after update to the new Geographic-lib-enchanced version

dkati, 2021-09-15:
i am on ROS melodic , ubuntu 18 and i cant catkin_make.i get
```
CMakeFiles/gps_mission.dir/src/gps_mission.cpp.o: In function `RobotLocalization::NavsatConversions::LLtoUTM(double, double, double&, double&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, double&)':
gps_mission.cpp:(.text+0x9f6): undefined reference to `GeographicLib::UTMUPS::Forward(double, double, int&, bool&, double&, double&, double&, double&, int, bool)'
gps_mission.cpp:(.text+0xa33): undefined reference to `GeographicLib::MGRS::Forward(int, bool, double, double, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)'
collect2: error: ld returned 1 exit status
ros_path_planner/CMakeFiles/gps_mission.dir/build.make:122: recipe for target '/home/dkati/thorvald_ws/devel/lib/ros_path_planner/gps_mission' failed
make[2]: *** [/home/dkati/thorvald_ws/devel/lib/ros_path_planner/gps_mission] Error 1
CMakeFiles/Makefile2:11919: recipe for target 'ros_path_planner/CMakeFiles/gps_mission.dir/all' failed
make[1]: *** [ros_path_planner/CMakeFiles/gps_mission.dir/all] Error 2
```
Cant find any solution

ayrton04, 2021-09-15:
Did you `rosdep install robot_localization`?

dkati, 2021-09-15:
yes i did

ayrton04, 2021-09-15:
The ROS build farm creates a Bionic container, and then clones everything, install the ROS dependencies, and then builds and tests the package. And that passes just fine with the latest commit to the `melodic-devel` branch:

https://build.ros.org/job/Mdev__robot_localization__ubuntu_bionic_amd64/

I have no problems building the head commit on the `melodic-devel` branch in Ubuntu 18.04.

Do you happen to have `r_l` installed from binaries as well as from source? What's the output when you do this?

```
cd /home/dkati/thorvald_ws/
rosdep install --from-paths src --ignore-src
```

dkati, 2021-09-15:
#All required rosdeps installed successfully

Thats the output. The fun fact is that it was working before 5 hours before i run apt-get update/upgrade.

ayrton04, 2021-09-15:
Ah, you are building your own package and are using `r_l` as a dependency. Your ticket sounded as if you couldn't build `r_l` itself.

My guess is the `geographiclib` dependencies aren't getting exported properly. You can try cloning the `r_l` repo into your workspace and see if you can work out what's missing in the CMakeLists.txt file. I won't have cycles just now to look into this. @Timple, any thoughts?

ayrton04, 2021-09-15:
https://github.com/cra-ros-pkg/robot_localization/blob/47b608256f08a0df5a1dd038c94fd5d071e16c72/CMakeLists.txt#L115

I think we need `GeographicLib` added to the `DEPENDS` list. If you can confirm that works, that would be great.

dkati, 2021-09-15:
I am out of office now. I will check tomorrow

dkati, 2021-09-20:
Problem persists since i realized that the DEPENDS list already contains the GeographicLib .

Reminder: I am on melodic branch ( https://github.com/cra-ros-pkg/robot_localization/commit/e9dfb75b5b6a2e53c00e47cc417d49cbad8b306d )

dkati, 2021-09-20:
FYI:
I managed to build my package by manually adding geographiclib dependency on cmake/package file.

So, shouldnt it be automatically imported while my package uses robot_localization as dependency?
aka shouldnt robot_localization expose that lib ?

ayrton04, 2021-09-20:
That is the purpose of this line:

https://github.com/cra-ros-pkg/robot_localization/blob/e9dfb75b5b6a2e53c00e47cc417d49cbad8b306d/CMakeLists.txt#L114

So the question is why isn't that line doing its job?

ayrton04, 2021-10-06:
Can you go to `<your workspace>/build/<your package>` and type `make VERBOSE=1` and see what build command is being issued?

melst33, 2021-10-14:
Here is the output:
make: *** No targets specified and no makefile found.  Stop.

> Any chance you can run the command I posted and show the output?

ayrton04, 2021-11-01:
@melst33, two things:

1. Go to your workspace directory (`/home/administrator/catkin_ws`). Type `rosdep install --from-paths src --ignore-src`. I see that you have another package that has missing headers, which makes me think you didn't run `rosdep install`.
2. Type `cd /home/administrator/catkin_ws/build/outdoor_waypoint_nav && make VERBOSE=1`

You are going to the `waypoint_nav` directory, but the one failing is `outdoor_waypoint_nav`. You are also (at least so far) having a different issue than the OP. That was a linking issue, whereas this is an include directory issue.

dkati, 2022-08-01:
bringing up this, i would like to report that despite the fact that, the commit (https://github.com/cra-ros-pkg/robot_localization/commit/e9dfb75b5b6a2e53c00e47cc417d49cbad8b306d) is merged, it is NEVER released on the melodic repository. When someone sudo apt-get install ros-melodic-robot-localization , it installed 2.6.11 where geographic lib commit is missing

ayrton04, 2022-08-03:
Yeah, just needs to be bloomed. I can do that soon.

ayrton04, 2022-08-08:
https://github.com/ros/rosdistro/pull/34089