cra-ros-pkg/robot_localization #696: Can I get Linear velocity from only IMU measurement?

astronaut71, 2021-09-23:
Hi 

I have IMU sensor that gives me the raw data such as orientation(in x,y,z) , Angular and Linear acceleration (also in x,y,z) So I have 9 axis IMU. Im using ROS and doing some Gazebo UUV simulation. Furthermore, I want to get linear velocity from the raw IMU data. If I do integration over time there will be accumulated error and will not be accurate with the time when for example the robot makes turns. So If I use

```
acceleration_x = (msg->linear_acceleration.x + 9.81 * sin(pitch)) * cos(pitch); 
acceleration_y = (msg->linear_acceleration.y - 9.81 * sin(roll)) * cos(roll);
```

So integrating linear acceleration is very bad,

`Velocity_x= Velocity_old_x+acceleration_x*dt;`

because integrates the acceleration without taking into account any possible rotation of the sensor, which means that the results will probably be terrible if the sensor rotates at all. Can this package provide accurate Linear Velocity?

ayrton04, 2021-09-23:
This question is better suited for answers.ros.org. Would you mind asking over there? And please be patient, as I don't get many cycles to dedicate to this package these days.

closed 2021-09-23 by ayrton04

astronaut71, 2021-09-26:
ok. So how about using this package, to configure it to use only IMU sensor  ( 9 DOF orientation, angular velocity and linear acceleration) as an input to obtain the  published topic o`dometry/filtered (nav_msgs/Odometry) ` as this topic contain  linear velocity in the twist message?

ayrton04, 2021-09-27:
Please ask questions on answers.ros.org.

astronaut71, 2021-09-27:
ok. please can you answer on answers.ros.org. Thanks

ayrton04, 2021-09-27:
Yes, when I have spare time. I'm behind on that, so I'll have to ask you to please be patient. Thanks.

astronaut71, 2021-09-27:
sure sure no problem. its this link https://answers.ros.org/question/387393/tf-is-broken-when-using-robot_localization-package-with-only-imu-sensor-and-how-correctly-to-use-and-configurate-the-package/

astronaut71, 2021-09-27:
TF looks ok now(although not 100% sure ) but Im not sure that I have done the correct configuration and launch file when only IMU sensor input is used (no odometer or any other sensor). Thanks

DoanNguyenTrong, 2022-08-05:
@astronaut71 Hi, can you please share with me how good you can achieve with this solution? What's happen with Odom after your robot runs for quite some time (let say 5hrs)?