cra-ros-pkg/robot_localization #704: ekf_node error in rolling version

figoowen2003, 2021-11-04:
## Bug report

**Required Info:**

- Operating System:
  - 
- Installation type:
  sudo apt install ros-rolling-robot-localization
- Version or commit hash:
  - 

#### Steps to reproduce issue
 I have installed the robot localization pacakage of rolling successfully. When I tried to run ekf_node, an error happened. The commands are,
sudo apt install ros-rolling-robot-localization
ros2 run robot_localization ekf_node

#### Expected behavior
 The ekf_node should be running
#### Actual behavior
 An error occured:
/opt/ros/rolling/lib/robot_localization/ekf_node: symbol lookup error: /opt/ros/rolling/lib/robot_localization/ekf_node: undefined symbol: _ZN6rclcpp4initEiPKPKcRKNS_11InitOptionsENS_20SignalHandlerOptionsE
#### Additional information
In the repo, I could not find the rolling branch.

ayrton04, 2022-02-07:
This looks like a linking issue with the `rclcpp::init` method. Did that method change for rolling?

SteveMacenski, 2022-02-07:
Yeah this is an issue where they updated / didn't link properly / source setup, not in the source code.

closed 2022-02-07 by SteveMacenski