cra-ros-pkg/robot_localization #707: can't subtract times with different time sources [1 != 2]

mnissov, 2021-11-22:
## Bug report
Running `robot_localization` ekf node returns error 
```bash
$ ros2 launch tractor_gazebo ekf.launch.py 
[INFO] [launch]: All log files can be found below /home/morten/.ros/log/2021-11-22-16-25-44-963334-agrirobot-123057
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ekf_node-1]: process started with pid [123062]
[ekf_node-1] X acceleration is being measured from IMU; X velocity control input is disabled
[ekf_node-1] terminate called after throwing an instance of 'std::runtime_error'
[ekf_node-1]   what():  can't subtract times with different time sources [1 != 2]
```
when used in conjunction with gazebo simulation.

Looking at `/odom` and `/imu/data` topics, timestamps match up 
[image]

In addition, all nodes are with `use_sim_time` : True

**EKF Launch File**:
```python
from launch import LaunchDescription
from ament_index_python.packages import get_package_share_directory
import launch_ros.actions
import os
import yaml
from launch.substitutions import EnvironmentVariable
import pathlib
import launch.actions
from launch.actions import DeclareLaunchArgument

def generate_launch_description():
    return LaunchDescription(
        [
            launch_ros.actions.Node(
                package="robot_localization",
                executable="ekf_node",
                name="ekf_filter_node",
                output="screen",
                parameters=[
                    {"use_sim_time": True},
                    os.path.join(
                        get_package_share_directory("tractor_gazebo"),
                        "config",
                        "ekf.yaml",
                    ),
                ],
            ),
        ]
    )

```

**EKF Params**:
```yaml
### ekf config file ###
ekf_filter_node:
    ros__parameters:
        frequency: 50.0
        sensor_timeout: 0.1
        two_d_mode: true
        transform_time_offset: 0.1
        transform_timeout: 0.0
        print_diagnostics: true
        publish_tf: true
        publish_acceleration: true
        map_frame: map              # Defaults to "map" if unspecified
        odom_frame: odom            # Defaults to "odom" if unspecified
        base_link_frame: base_link  # Defaults to "base_link" if unspecified
        world_frame: odom           # Defaults to the value of odom_frame if unspecified
        odom0: odom
        odom0_config: [false,  false,  false,
                       false, false, false,
                       true, true, true,
                       false, false, false,
                       false, false, false]
        odom0_queue_size: 2
        odom0_nodelay: false
        odom0_differential: false
        odom0_relative: false
        odom0_rejection_threshold: 2
        imu0: imu/data
        imu0_config: [false, false, false,
                      true,  true,  true,
                      false, false, false,
                      true,  true,  true,
                      true,  true,  true]
        imu0_nodelay: false
        imu0_differential: false
        imu0_relative: true
        imu0_queue_size: 5
        imu0_pose_rejection_threshold: 0.8                 # Note the difference in parameter names
        imu0_twist_rejection_threshold: 0.8                #
        imu0_linear_acceleration_rejection_threshold: 0.8
        imu0_remove_gravitational_acceleration: true
        use_control: true
        stamped_control: false
        control_timeout: 0.2
        control_config: [true, false, false, false, false, true]
        acceleration_limits: [1.3, 0.0, 0.0, 0.0, 0.0, 3.4]
        deceleration_limits: [1.3, 0.0, 0.0, 0.0, 0.0, 4.5]
        acceleration_gains: [0.8, 0.0, 0.0, 0.0, 0.0, 0.9]
        deceleration_gains: [1.0, 0.0, 0.0, 0.0, 0.0, 1.0]
```

**Required Info:**

- Operating System:
  - Ubuntu 20.04, ros2 Foxy
- Installation type:
  - installed `robot_localization` both from binary and source, both present same problem.

mnissov, 2021-11-22:
Just noticed, looking at solution provided to [#679](https://github.com/cra-ros-pkg/robot_localization/pull/679) the fix was changing 
`last_measurement_time_(0, 0, RCL_ROS_TIME), latest_control_time_(0),` to `last_measurement_time_(0, 0, RCL_ROS_TIME), latest_control_time_(0, 0, RCL_ROS_TIME),` in `src/filter_base.cpp`. This change has not been made in `foxy-devel`, see [link](https://github.com/cra-ros-pkg/robot_localization/blob/foxy-devel/src/filter_base.cpp#L50) so perhaps this could be the cause?

SteveMacenski, 2021-11-22:
Yeah please submit a PR

mnissov, 2021-11-24:
Solved by [#708](https://github.com/cra-ros-pkg/robot_localization/pull/708)

closed 2021-11-24 by mnissov