cra-ros-pkg/robot_localization #710: Yaw-offset and magnetic declination radiance calculation

Delbina, 2021-12-05:
Hi 
Does anybody to explain how to calculate yaw-offset and magnetic declination radiance?
I searched alot bet get more confused and found nothing 
Thanks

ayrton04, 2021-12-07:
In general, it's best to ask questions on answers.ros.org, but for now:

http://docs.ros.org/en/noetic/api/robot_localization/html/integrating_gps.html

closed 2021-12-07 by ayrton04

Delbina, 2021-12-07:
thank you for closing this issue,
but I am familiar with the link which you mentioned, I did not find the answer because of this asking here

ayrton04, 2021-12-07:
OK, then please use answers.ros.org, as all questions should be directed there.

Delbina, 2021-12-14:
hi, 
would you please answer my question in ROS answers:
https://answers.ros.org/question/392333/how-to-specify-yaw-offset-and-magnetic-declination-radiance-in-robot_localization-package/
thanks

ayrton04, 2021-12-15:
I will try to get to it soon. I don't get a lot of cycles to answer questions these days. I have a backlog of questions to answer, and I answer them in the order that they were asked.

Delbina, 2022-01-06:
Dear TOM,
would you please answer my question in ROS answers, I want to do the test, but I get stuck in this part for months!
https://answers.ros.org/question/392333/how-to-specify-yaw-offset-and-magnetic-declination-radiance-in-robot_localization-package/
thanks

ayrton04, 2022-01-07:
I'll try to get to it soon. I don't get a lot of time to support this package these days, but I'll try to carve out some ROS Answers time this week.