cra-ros-pkg/robot_localization #714: Changing the name of ‘ekf_node’ makes it stop functioning.

TaoYibo1866, 2022-01-11:
I am working on ros2-foxy, and installed this package through `sudo apt install ros-foxy-robot-localization`.
When i do `ros2 launch robot_localization ekf.launch.py`, everything is fine and `ros2 topic list` shows:
```
/cmd_vel
/diagnostics
/example/imu
/example/odom
/example/odom2
/example/pose
/example/twist
/odometry/filtered
/parameter_events
/rosout
/set_pose
/tf
/tf_static
```
However, if i modify the launch file into
```
# Copyright 2018 Open Source Robotics Foundation, Inc.
# Copyright 2019 Samsung Research America
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.

from launch import LaunchDescription
from ament_index_python.packages import get_package_share_directory
import launch_ros.actions
import os
import yaml
from launch.substitutions import EnvironmentVariable
import pathlib
import launch.actions
from launch.actions import DeclareLaunchArgument

def generate_launch_description():
    return LaunchDescription([
        launch_ros.actions.Node(
            package='robot_localization',
            executable='ekf_node',
            name='ekf_filter',
            output='screen',
            parameters=[os.path.join(get_package_share_directory("robot_localization"), 'params', 'ekf.yaml')],
           ),
])
```
which basically only rename the node from 'ekf_filter_node' to 'ekf_filter', the `ros2 topic list` shows:
```
/diagnostics
/odometry/filtered
/parameter_events
/rosout
/set_pose
/tf
/tf_static
```
This indicates that no subscribtion is done. Is this a bug or i just do something wrong?

ayrton04, 2022-01-11:
What does `ros2 topic list` show when you don't change the name? These topics are produced by the EKF:

```
/odometry/filtered
/set_pose
```

If `ros2 topic list` doesn't show your sensor data, then is it possible that data isn't being published?

ayrton04, 2022-01-11:
In any case, I think this should be asked on answers.ros.org or Robotics Stack Exchange.

TaoYibo1866, 2022-01-11:
When I don’t change the name the topic output is
```
/cmd_vel
/diagnostics
/example/imu
/example/odom
/example/odom2
/example/pose
/example/twist
/odometry/filtered
/parameter_events
/rosout
/set_pose
/tf
/tf_static
```

ayrton04, 2022-01-11:
That's because the parameters are being loaded for a node called `ekf_filter_node`:

https://github.com/cra-ros-pkg/robot_localization/blob/5d2e88268d988135af8642f4814cac7e2940cc87/params/ekf.yaml#L2

You need to update `ekf.yaml` so that line says `ekf_filter`.

ayrton04, 2022-01-11:
No worries!

closed 2022-01-11 by ayrton04