cra-ros-pkg/robot_localization #724: Change /fromLL and /toLL services or add new ones to accept and return arrays of points/geopoints

pepisg, 2022-02-14:
## Feature request

#### Feature description
Right now the `fromLL` and `toLL` services accept and return only one point / geopoint at a time, however in some applications it may be valuable to be able to process several points in a single service call rather than performing several successive calls with a single point each one. 
Thus it would be great if the services could be changed or new services could be included to receive an array of point or geopoints and return an array of their counterparts

#### Implementation considerations
I imagine this should not break the current services, as RL would stop working for people who already use them as they are today. New services definitions may look as these:

| fromLL                                                                        | toLL                                                                          |
|-------------------------------------------------------------------------------|-------------------------------------------------------------------------------|
| ```geographic_msgs/GeoPoint[]  ll_points --- geometry_msgs/Point[]   map_points``` | ```geometry_msgs/Point[]   map_points --- geographic_msgs/GeoPoint[]   ll_points``` |

If you agree this may be valuable I can work on the PR.

Thanks!

SteveMacenski, 2022-02-15:
Please submit a PR, happy to review

Arjan-TNO, 2022-02-25:
While you're on it: it would also be nice to go from `geometry_msgs/Point` to `geometry_msgs/PointStamped`. This allows the fromLL/toLL service to also pass the name of the `map_frame` into the message, as other ROS-nodes may not know this...